mu-editor/mu #1652: Crash Report

RogerArtemis, 2021-06-21:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_GB
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

```

RogerArtemis, 2021-06-21:
Traceback (most recent call last):
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
Next steps:

RogerArtemis, 2021-06-21:
These errors said "Could not quarantine", they also seemed to prevent me from deleting the programme.

carlosperate, 2021-06-21:
Btw, that path expanded would be `C:\Users\crbre\AppData\Local\python\mu\Logs`

RogerArtemis, 2021-06-22:
Please find attached the “Crash Report” for MU programme altercation.
 
Mu Crash Report
Something has gone wrong with Mu. Here's the error:
Traceback (most recent call last):
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\crbre\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip
Next steps:
• Submit a new bug report (you will need a GitHub account for this to work).
• Please attach your log file to the bug report. Your log file can be found here: C:\Users\crbre\AppData\Local\python\mu\Logs\mu.log
• Be prepared to answer questions we may have about the bug report so we have all the information we need so we can fix it!
Apologies for the inconvenience and thanks for the help.

From: Carlos Pereira Atencio
Sent: 21 June 2021 17:22
To: mu-editor/mu
Cc: RogerArtemis; Mention
Subject: Re: [mu-editor/mu] Crash Report (#1652)

Btw, that path expanded would be C:\Users\crbre\AppData\Local\python\mu\Logs

carlosperate, 2021-06-22:
Hi Rogers,
Can you also send us the log file from C:\Users\crbre\AppData\Local\python\mu\Logs?
If you tried to add it as an attachment to your previous email unfortunately the files are not forwarded to GitHub, to upload the file you need to do it via the GitHub website (you can drag&drop the file into a response text box): https://github.com/mu-editor/mu/issues/1652
Thanks!

carlosperate, 2021-08-06:
Hi @RogerArtemis, as this issue is covered in https://github.com/mu-editor/mu/issues/1532 I'm merging this issue over there: https://github.com/mu-editor/mu/issues/1532#issuecomment-894179118

closed 2021-08-06 by carlosperate