mu-editor/mu #1655: New to programming, pressed REPL for Python3 and it crashed

ronsenyor, 2021-06-22:
# Bug Report

I was trying to launch REPL after installing Mu editor via Python3 on my ChromeOSbook. 

Below is the error message:

Traceback (most recent call last):
  File "/home/ronsenyor/.local/lib/python3.7/site-packages/mu/modes/python3.py", line 372, in on_kernel_start
    self.view.add_jupyter_repl(kernel_manager, kernel_client)
  File "/home/ronsenyor/.local/lib/python3.7/site-packages/mu/interface/main.py", line 581, in add_jupyter_repl
    kernel_client.start_channels()
  File "/home/ronsenyor/.local/lib/python3.7/site-packages/qtconsole/kernel_mixins.py", line 47, in start_channels
    super(QtKernelClientMixin, self).start_channels(*args, **kw)
  File "/home/ronsenyor/.local/lib/python3.7/site-packages/jupyter_client/threaded.py", line 248, in start_channels
    super().start_channels(shell, iopub, stdin, hb, control)
  File "/home/ronsenyor/.local/lib/python3.7/site-packages/jupyter_client/client.py", line 295, in start_channels
    self.hb_channel.start()
  File "/home/ronsenyor/.local/lib/python3.7/site-packages/jupyter_client/client.py", line 364, in hb_channel
    self._hb_channel = self.hb_channel_class(self.context, self.session, url)
  File "/home/ronsenyor/.local/lib/python3.7/site-packages/qtconsole/util.py", line 65, in __new__
    QClass.__init__(inst)
TypeError: __init__() missing 1 required positional argument: 'context'

carlosperate, 2021-06-23:
Hi @ronsenyor, thanks for the report!

What command exactly did you use to install the mu-editor?

ronsenyor, 2021-06-23:
> Hi @ronsenyor, thanks for the report!
> 
> What command exactly did you use to install the mu-editor?

Thanks for offering to help

- I installed pip, sudo apt-get install python3-pip
- then I ran
pip3 install mu-editor

And then I was able to start up mueditor with 

python3 -m mu

And I ran into the issue when I pressed REPL

carlosperate, 2021-06-23:
Great, thanks!
Could you run this command in a terminal and past the output here?

```
pip3 list
```

ronsenyor, 2021-06-23:
> Great, thanks!
> Could you run this command in a terminal and past the output here?
> 
> ```
> pip3 list
> ```

Thanks Carlos

Package                Version 
---------------------- --------
adafruit-board-toolkit 1.1.0   
appdirs                1.4.4   
asn1crypto             0.24.0  
backcall               0.2.0   
black                  21.6b0  
click                  8.0.1   
cryptography           2.6.1   
decorator              5.0.9   
distlib                0.3.2   
entrypoints            0.3     
filelock               3.0.12  
flake8                 3.9.2   
importlib-metadata     4.5.0   
ipykernel              5.5.5   
ipython                7.24.1  
ipython-genutils       0.2.0   
jedi                   0.18.0  
jupyter-client         6.1.13  
jupyter-core           4.7.1   
keyring                17.1.1  
keyrings.alt           3.1.1   
matplotlib-inline      0.1.2   
mccabe                 0.6.1   
mu-editor              1.1.0b4 
mypy-extensions        0.4.3   
nest-asyncio           1.5.1   
nudatus                0.0.5   
parso                  0.8.2   
pathspec               0.8.1   
pexpect                4.8.0   
pickleshare            0.7.5   
pip                    18.1    
prompt-toolkit         3.0.19  
ptyprocess             0.7.0   
pycodestyle            2.7.0   
pycrypto               2.6.1   
pyflakes               2.3.1   
Pygments               2.9.0   
PyGObject              3.30.4  
PyQt5                  5.13.2  
PyQt5-sip              12.9.0  
PyQtChart              5.13.1  
pyserial               3.5     
python-dateutil        2.8.1   
pyxdg                  0.25    
pyzmq                  22.1.0  
QScintilla             2.11.3  
qtconsole              4.7.7   
QtPy                   1.9.0   
regex                  2021.4.4
SecretStorage          2.3.1   
semver                 2.13.0  
setuptools             40.8.0  
six                    1.12.0  
toml                   0.10.2  
tornado                6.1     
traitlets              5.0.5   
typed-ast              1.4.3   
typing-extensions      3.10.0.0
virtualenv             20.4.7  
wcwidth                0.2.5   
wheel                  0.32.3  
zipp                   3.4.1

carlosperate, 2021-06-23:
Thanks @ronsenyor!
Looks like you might have been affected by this issue with `jupyter-client`: https://github.com/mu-editor/mu/issues/1444
Version  6.1.13 was only available for less than a day before it was removed from PyPI, if you've installed Mu editor recently I guess something else must have installed it around the 7th of April.

Could you try running this command and then launching Mu again? 
```
pip3 install jupyter-client==6.1.12
```

ronsenyor, 2021-06-23:
> Thanks @ronsenyor!
> Looks like you might have been affected by this issue with `jupyter-client`: #1444
> Version 6.1.13 was only available for less than a day before it was removed from PyPI, if you've [...]

REPL WORKED!!!! Thank you so much Carlos. Wow. Thank you.

carlosperate, 2021-06-23:
Awesome, I'm really glad we got it working! 🎉 

I'll close this issue as resolved, thanks for testing and don't hesitate to open more issues if you find any other problems 👍

closed 2021-06-23 by carlosperate