mu-editor/mu #1656: Crash Report

noahmmreh, 2021-06-23:
Please I don't what happened to my Mu editor.
It cannot open, please can anone help me?
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Noah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Noah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\Noah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\Noah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```C:\Users\Noah\AppData\Local\python\mu\Logs\mu.log

carlosperate, 2021-06-23:
Thanks for the report @noahmmreh!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Noah\AppData\Local\python\mu\Logs\mu.log`

noahmmreh, 2021-06-23:
Mu Crash Report

Something has gone wrong with Mu. Here's the error:

Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Local\Programs\Mu
Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Noah\AppData\Local\Programs\Mu
Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Noah\AppData\Local\Programs\Mu
Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495,
in ensure_and_create
    self.ensure()
  File "C:\Users\Noah\AppData\Local\Programs\Mu
Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524,
in ensure
    self.ensure_key_modules()
  File "C:\Users\Noah\AppData\Local\Programs\Mu
Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602,
in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

Next steps:

   - Submit a new bug report
   <https://github.com/mu-editor/mu/issues/new?title=Crash%20Report&body=**REMEMBER%20TO%20ATTACH%20YOUR%20LOG%20FILE**%0A%0AVersion%3A%201.1.0.beta.4%0ALocale%3A%20en_US%0APlatform%3A%20Windows%2010%2010.0.19041%20AMD64%0A%0A%60%60%60%0ATraceback%20(most%20recent%20call%20last)%3A%0A%20%20File%20%22C%3A%5CUsers%5CNoah%5CAppData%5CLocal%5CPrograms%5CMu%20Editor%5CPython%5Clib%5Csite-packages%5Cmu%5Capp.py%22%2C%20line%20161%2C%20in%20run%0A%20%20%20%20raise%20ex%0A%20%20File%20%22C%3A%5CUsers%5CNoah%5CAppData%5CLocal%5CPrograms%5CMu%20Editor%5CPython%5Clib%5Csite-packages%5Cmu%5Capp.py%22%2C%20line%20148%2C%20in%20run%0A%20%20%20%20venv.ensure_and_create(self.display_text)%0A%20%20File%20%22C%3A%5CUsers%5CNoah%5CAppData%5CLocal%5CPrograms%5CMu%20Editor%5CPython%5Clib%5Csite-packages%5Cmu%5Cvirtual_environment.py%22%2C%20line%20495%2C%20in%20ensure_and_create%0A%20%20%20%20self.ensure()%0A%20%20File%20%22C%3A%5CUsers%5CNoah%5CAppData%5CLocal%5CPrograms%5CMu%20Editor%5CPython%5Clib%5Csite-packages%5Cmu%5Cvirtual_environment.py%22%2C%20line%20524%2C%20in%20ensure%0A%20%20%20%20self.ensure_key_modules()%0A%20%20File%20%22C%3A%5CUsers%5CNoah%5CAppData%5CLocal%5CPrograms%5CMu%20Editor%5CPython%5Clib%5Csite-packages%5Cmu%5Cvirtual_environment.py%22%2C%20line%20602%2C%20in%20ensure_key_modules%0A%20%20%20%20raise%20VirtualEnvironmentEnsureError(%0Amu.virtual_environment.VirtualEnvironmentEnsureError%3A%20Failed%20to%20import%3A%20pgzero%0A%0A%60%60%60>
(you
   will need a GitHub account <https://github.com/> for this to work).
   - *Please attach your log file to the bug report.* Your log file can be
   found here: C:\Users\Noah\AppData\Local\python\mu\Logs\mu.log
   - Be prepared to answer questions we may have about the bug report so we
   have all the information we need so we can fix it!

Apologies for the inconvenience and thanks for the help.

carlosperate, 2021-06-23:
Thank you @noahmmreh! 

For this problem we really need the Mu logs to be able to help you.
Could you navigate to this folder on your computer `C:\Users\Noah\AppData\Local\python\mu\Logs\` and attach the `mu.log` to this GitHub issue? Link to the issue: https://github.com/mu-editor/mu/issues/1656

For adding the file you'll have to use the GitHub website, unfortunately if you attach the file in an email it doesn't reach us. Thanks!

carlosperate, 2021-06-23:
Thank you @noahmmreh, we already have that error, but unfortunately it's not enough to figure out the problem,
We need the Mu log, can you find the log file in `C:\Users\Noah\AppData\Local\python\mu\Logs\mu.log` and post it on this GitHub issue (an email attachment won't work).
https://github.com/mu-editor/mu/issues/1656

noahmmreh, 2021-06-23:
I don't know GitHub and I can't do that.

carlosperate, 2021-06-23:
If you are able to find the file in your computer all you have to do is go to this URL: https://github.com/mu-editor/mu/issues/1656

And the drag and drop the file into the text editor:

[image]

carlosperate, 2021-06-24:
Closing this issue as it has been reported as working in https://github.com/mu-editor/mu/issues/1662#issuecomment-867543609.
Thanks for the report!

closed 2021-06-24 by carlosperate