mu-editor/mu #1658: Crash Report

Tozaki31, 2021-06-23:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: ja_JP
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\tozak\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\tozak\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\tozak\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\tozak\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\tozak\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

carlosperate, 2021-06-23:
Thanks for the report @Tozaki31!
Would you be able to attach the Mu log to this GH Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\tozak\AppData\Local\python\mu\Logs`

carlosperate, 2021-08-12:
Hi @Tozaki31,

We can continue the debugging of this issue in https://github.com/mu-editor/mu/issues/1646 👍

closed 2021-08-12 by carlosperate