mu-editor/mu #1659: I got the same message with a Micro:bit V2 on Windows 8.1.  I had upgraded Mu to the lastest version (64-bit).  I also tried on Windows 10 and just used the older version of Mu because my Windows 10 was 32-bit.  I did not get the message on the older Mu.  On Windows 8.1, which is 64-bit, with the new version on Mu, I not only got the "Could not find an attached BBC Micro:bit" and there was a red "X" in the lower right to indicate this.  However, Windows 10 was recognizing the Micro:bit on usb, and I was in fact, able to copy a .hex file to the Micro:bit usb drive and it worked ran on the Micro.bit.  I am not sure what gives with the new 64-bit Mu and recognizing the Micro:bit on usb.  Just another note, the web version on micropython works fine on the same Microbit V2.

carlosperate, 2021-06-23:
I got the same message with a Micro:bit V2 on Windows 8.1.  I had upgraded Mu to the lastest version (64-bit).  I also tried on Windows 10 and just used the older version of Mu because my Windows 10 was 32-bit.  I did not get the message on the older Mu.  On Windows 8.1, which is 64-bit, with the new version on Mu, I not only got the "Could not find an attached BBC Micro:bit" and there was a red "X" in the lower right to indicate this.  However, Windows 10 was recognizing the Micro:bit on usb, and I was in fact, able to copy a .hex file to the Micro:bit usb drive and it worked ran on the Micro.bit.  I am not sure what gives with the new 64-bit Mu and recognizing the Micro:bit on usb.  Just another note, the web version on micropython works fine on the same Microbit V2.

_Originally posted by @jnank1 in https://github.com/mu-editor/mu/issues/513#issuecomment-866755510_

carlosperate, 2021-06-23:
Thanks for the report @jnank1!
Could you follow these steps to help us diagnose the issue?
- Open Mu v1.1.0-beta.4
- Go into the micro:bit mode
- Connect the micro:bit to the computer via USB
- Wait a few seconds
- Open any micro:bit python file in Mu
- Press the flash button
- Get the Mu logs and add them to this GitHub issue
    - Info on how to get the logs https://codewith.mu/en/howto/1.1/read_logs

The logs after following these steps will contain a lot of useful information for us, thanks!

jnank1, 2021-06-23:
The following Mu v1.10-beta.4 logs are from a session on windows 8.1 (64-bit) with a Micro:bit v2. on 6/23/21 at about 5:45 local time (US, EDT).  It came up with a picro:bit python file loaded, one that I knew ran correctly on the V2 since I had used the web version to load it, but a different program was currently loaded on the micro:bit.  Mu is in micro:bit mode.  The red X in showing in the lower right, indicating Mu is not connected to the micro:bit.  I click on "files" and "REPL" .  Both return the " Could not find an attached BBC Micro:bit" message.  When I pressed "flash" , Mu began downloading the file to the Micro:bit.  At this point I remember this happening before, but it did not finish correctly and now I realize it was because the code had an error in it.  This time the download completed correctly and the code ran on the Micro:bit the way it is supposed  to.  Note,  I want to say also that initially (couple days ago) I did get a 537 (Sad face) error a couple other times I tried to flash on this system.  I think that was probably a different problem, because I changed cables and have not got it again.  But on the other hand I could not see how it could have been a cable problem because the other problems persists, Red X, "Could not find attached....."  Oh well, I hope this logs tells you something. 

2021-06-23 17:32:45,481 - root:265(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.4
2021-06-23 17:32:45,512 - root:266(run) INFO: uname_result(system='Windows', node='Win8_Desktop', release='8.1', version='6.3.9600', machine='AMD64', processor='Intel64 Family 6 Model 23 Stepping 10, GenuineIntel')
2021-06-23 17:32:45,512 - root:267(run) INFO: Python path: ['C:\\Users\\jnank\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\jnank\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\jnank\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\jnank\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\jnank\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\jnank\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\jnank\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\jnank\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\jnank\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\jnank\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\jnank\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\jnank\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-06-23 17:32:45,512 - root:268(run) INFO: Language code: en_US
2021-06-23 17:32:46,688 - mu.virtual_environment:480(ensure_and_create) INFO: Added log handler.
2021-06-23 17:32:47,126 - mu.virtual_environment:491(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-06-23 17:32:47,126 - mu.virtual_environment:542(ensure_path) INFO: Virtual Environment found at: C:\Users\jnank\AppData\Local\python\mu\mu_venv-38-20210621-091605
2021-06-23 17:32:47,141 - mu.virtual_environment:555(ensure_interpreter) INFO: Interpreter found at: C:\Users\jnank\AppData\Local\python\mu\mu_venv-38-20210621-091605\scripts\python.exe
2021-06-23 17:32:49,059 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 38
2021-06-23 17:32:49,059 - mu.virtual_environment:585(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-06-23 17:32:49,074 - mu.virtual_environment:611(ensure_pip) INFO: Pip found at: C:\Users\jnank\AppData\Local\python\mu\mu_venv-38-20210621-091605\scripts\pip.exe
2021-06-23 17:32:49,090 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-06-23 17:32:49,253 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-23 17:32:49,253 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: flask
2021-06-23 17:32:50,970 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-23 17:32:50,970 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: qtconsole
2021-06-23 17:32:51,380 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-23 17:32:51,380 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-06-23 17:32:51,797 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-23 17:32:51,797 - mu.virtual_environment:597(ensure_key_modules) DEBUG: Verifying import of: adafruit_board_toolkit
2021-06-23 17:32:51,983 - mu.virtual_environment:379(run_subprocess) DEBUG: Process returned 0; output: 
2021-06-23 17:32:51,983 - mu.virtual_environment:496(ensure_and_create) INFO: Valid virtual environment found at C:\Users\jnank\AppData\Local\python\mu\mu_venv-38-20210621-091605
2021-06-23 17:32:51,983 - mu.settings:169(save) DEBUG: Saving to C:\Users\jnank\AppData\Local\python\mu\venv.json
2021-06-23 17:32:51,998 - mu.virtual_environment:509(ensure_and_create) DEBUG: Emitter: <bound PYQT_SIGNAL display_text of StartupWorker object at 0x000000691FA8AA60>; Splash Handler; <SplashLogHandler (INFO)>
2021-06-23 17:32:52,349 - mu.logic:758(__init__) INFO: Setting up editor.
[... cut ...]
    sleep(400)
    display.clear()
while m < 9:
[... 296 lines cut ...]

pywin32                300

pyzmq                  22.0.3

qtconsole              4.7.4

QtPy                   1.9.0

reedsolo               1.5.4

setuptools             56.0.0

six                    1.15.0

tornado                6.1

traitlets              5.0.5

wcwidth                0.2.5

Werkzeug               1.0.1

wheel                  0.36.2
2021-06-23 17:48:24,054 - mu.virtual_environment:797(installed_packages) INFO: []

microbit-mark, 2021-06-24:
> https://support.microbit.org/a/tickets/46124

Originally reported to the Foundation on support. Adding the link here so I can tie the threads together.

carlosperate, 2021-08-24:
Thanks for the report @jnank1! 
Do you still have this problem?
If you do, can you try flashing the micro:bit with the Python editor first, and see if that helps? https://python.microbit.org

jnank1, 2021-08-25:
The Micro-bit already had micropython loaded and I could run programs.  The computer was recognizing micro-bit as storage so I could load the micropython file and use the Mu editor and then I could copy the program to the microbit and it would run immediately, but I could not use interactive function or save programs directly to the micro-bit from Mu.  The OS would recognize the micro-bit as storage but Mu did not.
Jeff Nank

jnank1, 2021-08-25:
I guess I could reload micropython (flash) instead of loading the micropython file, the file with a special extension,  I can't remember the name of it right now.

carlosperate, 2021-08-25:
Thanks @jnank1 it sounds like Windows didn't install the serial drivers automatically. Windows 8.1 should have automatically installed this driver from the internet the first time a micro:bit was connected, but maybe this didn't happen for some reason.

The way Mu flashes the micro:bit is that it first try to send the files via serial, and it all fails it then tries to use an old deprecated method to flash a hex file, this could explain the difference from the different Mu versions.

If you open the windows device manager (https://support.microsoft.com/en-us/windows/open-device-manager-a7f2db46-faaf-24f0-8b7b-9e4a6032fc8c), does it show any device without drivers? It should have like a small yellow warning triangle in the icon.

When flashing via the online editor (https://python.microbit.org) did you download a hex file and then copied it to the MICROBIT drive? Or did you use the WebUSB functionality and connected directly from the browser?