mu-editor/mu #1662: Crash Report

noahmmreh, 2021-06-24:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Noah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Noah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Noah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "C:\Users\Noah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "C:\Users\Noah\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

Mu is not open and I don't why

carlosperate, 2021-06-24:
Duplicated of https://github.com/mu-editor/mu/issues/1656.

closed 2021-06-24 by carlosperate

noahmmreh, 2021-06-24:
Mu is opened now
Thank you so much
But I am facing another issue with repl.it and the issue is that the share
or submit button is missing on repl.it.
Can you help me please?

carlosperate, 2021-06-24:
I haven't really used repl.it, sorry. I'd recommend checking if there is a way to contact support from your account.