mu-editor/mu #1664: Does not recognize 'walrus operator'

Dani-Long, 2021-06-25:
### What were you trying to do?

Use the walrus operator

### What steps did you take to trigger the issue?

The line of code is:
scrubfile := (open ('scrubfile.txt', 'at')

### What actually happened?

When checking the code:
_Whitespace before ':'_
_Whitespace after ':'_
_Missing whitespace around operator_

### Operating System Version

Debian GNU/Linux 10 (buster)

### Mu Version

1.0.3

### Other Info

[image]

### Editor Log

I keep getting an error message here that the body is too long!?!?!?!

carlosperate, 2021-06-25:
Thanks for the report @Dani-Long!

> I keep getting an error message here that the body is too long!?!?!?!

Apologies for that, the GitHub issue forms are new, but I believe the problem might be experiencing applies to normal issues as well, as there is a 64 KB of text limit.
You should be able to drag & drop the log file into a response to this issue, in Linux it should be located in `~/.cache/mu/mu.log`

How did you install Mu? What version of Python are you using? It's possible the version of the code checkers is older than the introduction of the walrus operator in Python 3.8. Nevertheless the code should still work as long as you are using Python 3.8+.

carlosperate, 2021-08-17:
Hi @Dani-Long,

Would you be able to check if beta.5 works for you and let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5
`pip install mu-editor==1.1.0b5`

Thanks!

carlosperate, 2021-09-21:
I'll close this due to lack of activity.
I tested this in beta.3 and seems to work fine, so it must have been fixed in a previous release.

Thanks for the report @Dani-Long, please let us know if you are still having this issue or any other!