mu-editor/mu #1665: mu-editor-1.1.0 beta 4

winston154, 2021-06-26:
### What were you trying to do?

installing mu-editor beta 4

### What steps did you take to trigger the issue?

python run.py

### What did you expect to happen?

mu-editor to run

### Operating System Version

fedora 35

### Mu Version

mu-editor 1.0.3

### Editor Log

Traceback (most recent call last):
  File "/home/chadhogue/Downloads/mu-1.1.0-beta.4/mu/app.py", line 161, in run
    raise ex
  File "/home/chadhogue/Downloads/mu-1.1.0-beta.4/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/chadhogue/Downloads/mu-1.1.0-beta.4/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/home/chadhogue/Downloads/mu-1.1.0-beta.4/mu/virtual_environment.py", line 623, in create
    self.create_venv()
  File "/home/chadhogue/Downloads/mu-1.1.0-beta.4/mu/virtual_environment.py", line 654, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python at /home/chadhogue/.local/share/mu/mu_venv-310-20210626-100011

carlosperate, 2021-06-26:
Thanks for the report @winston154!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

What steps exactly did you follow to install Mu? What version of Python and pip are you using?

winston154, 2021-06-26:
[mu.log](https://github.com/mu-editor/mu/files/6720788/mu.log)

I'm running Fedora 35 Alpha and Python 3.10
I used the following steps from https://mu.readthedocs.io/en/latest/ :

Mu works with Python 3.6 or above.

Clone the repository:

git clone https://github.com/mu-editor/mu.git
(Recommended) Upgrade local pip:

pip install --upgrade pip
Make a virtualenv, then install the requirements:

pip install -e ".[dev]"
Start Mu:

python run.py
Run the test suite:

make check
Read on to learn more about Mu, its aims and how you can contribute.

carlosperate, 2021-06-27:
Thanks @winston154!

If you are using a relatively modern version of pip, specially if you've updated it as indicated in those steps, it should have not let you installed Mu with Python 3.10. Due to some dependencies we are currently locked at versions 3.5 to 3.8:
https://github.com/mu-editor/mu/blob/23532e734fce719e8d3108ebab0ff78cc88f136f/setup.py#L112

Can you try with a compatible Python version?

carlosperate, 2021-08-12:
Hi @winston154, were you able to try to install Mu with a Python version below 3.9?

winston154, 2021-08-13:
Sorry for the delay getting back to you, I broke my Fedora Rawhide system so I reinstalled Fedora and I'll see about installing this tonight. Thanks for your help @carlosperate and I will let you know

carlosperate, 2021-08-13:
No worries at all, than you very much for all your help debugging this issue!

winston154, 2021-09-21:
I was not able to install beta 4 (now beta 5) but was able to install version I believe 1.00.3. I will have to try again to see what error codes I'm coming up with.

carlosperate, 2021-12-17:
Hi @winston154, are you still experiencing this issue with the latest beta.6? 
https://github.com/mu-editor/mu/releases/

winston154, 2021-12-18:
My computer died and I have not been able to try. My wife bought me a new one for Christmas and will have to wait till next weekend to hopefully try.
Thanks for checking

carlosperate, 2021-12-18:
Oh, I'm sorry to hear your computer died, let us know if you have a chance to try it once you get the new one. Happy holidays!

winston154, 2021-12-24:
OK, we celebrated Christmas today as my kids are off work and it's just easier. installed Fedora35, updated and rebooted. Now time to install Mu-editor. I enter the following command directly from MU-editor's website, it tells me I have to install pip3 and a handful of other dependencies, then I get the following, same as before:

$ pip3 install mu-editor==1.1.0b7
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement mu-editor==1.1.0b7 (from versions: 1.0.0b12, 1.0.0b13, 1.0.0b14, 1.0.0b15, 1.0.0b16, 1.0.0b17, 1.0.0rc1, 1.0.0, 1.0.1, 1.0.2, 1.0.3)
ERROR: No matching distribution found for mu-editor==1.1.0b7

going to keep trying as I'm sure its something simple I'm missing

carlosperate, 2021-12-24:
What version of Python are you using? The current verson of Mu is only compatible with Python 3.5 to 3.8.

winston154, 2021-12-24:
$ python3 --version
Python 3.10.1

Well, I guess that explains a lot! I assume I can downgrade to version 3.8?

carlosperate, 2021-12-27:
I'm not familiar with Fedora, so I don't know how easy it might be to downgrade or install other Python versions, but if you manage to install Python 3.8 that should work.