mu-editor/mu #1667: Crash Report

DLadyE, 2021-06-27:
### What were you trying to do?

install Mu Editor

### What steps did you take to trigger the issue?

I installed Mu Editor.. I ran it after the install was done and then this

### What did you expect to happen?

I would b able to use the program

### What actually happened?

this Crash report happened

### Operating System Version

Windows 10

### Mu Version

1.10

### Editor Log

Traceback (most recent call last):
  File "C:\Users\D\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\D\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\D\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\D\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\D\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

carlosperate, 2021-06-27:
Thanks for the report @DLadyE!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\D\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-08-06:
Hi @DLadyE, as this issue is covered in https://github.com/mu-editor/mu/issues/1532 I'm merging this issue over there: https://github.com/mu-editor/mu/issues/1532#issuecomment-894179642

closed 2021-08-06 by carlosperate