mu-editor/mu #1670: Crash Report

Quillsxi, 2021-06-27:
[mu.log](https://github.com/mu-editor/mu/files/6722329/mu.log)

Version: 1.1.0.beta.4
Locale: en_US
Platform: Linux 5.4.0-77-generic #86~18.04.1-Ubuntu SMP Fri Jun 18 01:23:22 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/dani/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 730, in install_baseline_packages
    wheels.download(interpreter=self.interpreter, logger=logger)
  File "/home/dani/.local/lib/python3.6/site-packages/mu/wheels/__init__.py", line 138, in download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download esptool==3.*

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/dani/.local/lib/python3.6/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/dani/.local/lib/python3.6/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/dani/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/home/dani/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 625, in create
    self.install_baseline_packages()
  File "/home/dani/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 732, in install_baseline_packages
    raise VirtualEnvironmentCreateError(exc.message)
mu.virtual_environment.VirtualEnvironmentCreateError: Pip was unable to download esptool==3.*

```

carlosperate, 2021-06-27:
Thanks for the report and the log @Quillsxi!

The issue seems to be that when installing the dependencies in the virtual environment it tries to build the `cryptography` package from source:

```
Collecting cryptography>=2.1.4 (from esptool==3.*)
  Using cached https://files.pythonhosted.org/packages/9b/77/461087a514d2e8ece1c975d8216bc03f7048e6090c5166bc34115afdaa53/cryptography-3.4.7.tar.gz
  Saved ./.local/lib/python3.6/site-packages/mu/wheels/cryptography-3.4.7.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-lkhcnpl2/cryptography/setup.py", line 14, in <module>
        from setuptools_rust import RustExtension
    ModuleNotFoundError: No module named 'setuptools_rust'
            =============================DEBUG ASSISTANCE==========================
            If you are seeing an error here please try the following to
            successfully install cryptography:
            Upgrade to the latest pip and try again. This will fix errors for most
            users. See: https://pip.pypa.io/en/stable/installing/#upgrading-pip
            =============================DEBUG ASSISTANCE==========================
    ----------------------------------------
```

What version of pip do you have installed?
Are you running Ubuntu 18.04.1?

The odd thing here is that technically this shouldn't be happening because pip inside the virtual environment is updated before trying to install this, and the latest version of pip should pick up a pre-built wheel from PyPI instead of trying to build it from source.

Can you upgrade pip and try again?

Quillsxi, 2021-06-27:
I am not on Ubuntu
My OS is Zorin OS 15.3
I tried updating pip and it did not work

carlosperate, 2021-06-28:
Thank you @Quillsxi! 

Can you try these command and copy/paste the result here?

```
pip3 --version
```

```
python3 -m pip --version
```

```
python3 -m pip install cryptography==3.4.7 -vv
```

abixadamj, 2021-06-28:
@Quillsxi - please try https://github.com/mu-editor/mu/discussions/1426#discussioncomment-897646

I know Zorin is based on Ubuntu, but I don't remember exactly n which version.

carlosperate, 2021-08-10:
Hi @Quillsxi, were you able to resolve this issue?
If not, could you try installing beta.5?

Quillsxi, 2021-09-25:
> @Quillsxi <https://github.com/Quillsxi> have you been able to test this
> with beta.5? There has been some changes in this area that we think could
> have helped with this issue:
> https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

sorry dude I don't mu anymore

Quillsxi, 2021-09-25:
Sorry for the late reply. I don't use Mu anymore

carlosperate, 2021-09-26:
No worries and thanks for the update!
I think this error specifically has been fixed in beta.5, so we can close this issue safely.
Thanks again for the report!

closed 2021-09-26 by carlosperate