mu-editor/mu #1671: Crash Report

luw-k, 2021-06-28:
never coded before this is probably very obvious. just downlaoding the Mu installer and can't get past this.. any ideas?

[mu.log](https://github.com/mu-editor/mu/files/6725355/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_GB
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\kate1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 283, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\kate1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\kate1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\kate1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\kate1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 626, in create
    self.register_baseline_packages()
  File "C:\Users\kate1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 751, in register_baseline_packages
    packages = list(self.pip.installed())
  File "C:\Users\kate1\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 289, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages

```

carlosperate, 2021-07-05:
Thanks for the report and the log @luw-k!
We've just released a new beta version of Mu, would you mind installing this version and trying again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

closed 2021-07-10 by luw-k

carlosperate, 2021-07-10:
Did the new version work @luw-k ?

luw-k, 2021-07-13:
hi sorry yes it did! thanks so much for your help really appreciate it!

carlosperate, 2021-07-13:
Awesome, thanks for the update! Really glad the new version fixes your issue.