mu-editor/mu #1672: Crash Report

brunixmnt, 2021-06-28:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Linux 5.8.0-59-generic #66~20.04.1-Ubuntu SMP Thu Jun 17 11:14:10 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 730, in install_baseline_packages
    wheels.download(interpreter=self.interpreter, logger=logger)
  File "/usr/local/lib/python3.8/dist-packages/mu/wheels/__init__.py", line 137, in download
    raise WheelsDownloadError(
mu.wheels.WheelsDownloadError: Pip was unable to download pgzero>=1.2.1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 161, in run
    raise ex
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 625, in create
    self.install_baseline_packages()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 732, in install_baseline_packages
    raise VirtualEnvironmentCreateError(exc.message)
mu.virtual_environment.VirtualEnvironmentCreateError: Pip was unable to download pgzero>=1.2.1

```

carlosperate, 2021-06-28:
Thanks for the report @brunixmnt!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

carlosperate, 2021-08-12:
Hi @brunixmnt,

Are you able to attach the log? File should be in `~/.cache/mu/`.

Would you also be able to check if beta.5 has resolved the issue let us know the result?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5
`pip install mu-editor==1.1.0b5`

Thanks!

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate