mu-editor/mu #1675: Informe de bloqueo

Roshelys, 2021-07-01:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_CA
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Roshelys Gondola\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Roshelys Gondola\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Roshelys Gondola\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\Roshelys Gondola\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 624, in create
    self.upgrade_pip()
  File "C:\Users\Roshelys Gondola\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 671, in upgrade_pip
    raise VirtualEnvironmentCreateError("Unable to upgrade pip")
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to upgrade pip

```

carlosperate, 2021-07-01:
Thanks for the report @Roshelys!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Roshelys Gondola\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-08-06:
Hi @Roshelys, as this issue is covered in https://github.com/mu-editor/mu/issues/1532 I'm merging this issue over there: https://github.com/mu-editor/mu/issues/1532#issuecomment-894180397

closed 2021-08-06 by carlosperate