mu-editor/mu #1677: Crash Report from Windows Installer

carlosperate, 2021-07-02:
**REMEMBER TO ATTACH YOUR LOG FILE**

[mu.log](https://github.com/mu-editor/mu/files/6754357/mu.log)

Version: 1.1.0.beta.4
Locale: en_US
Platform: Windows 10 10.0.17763 AMD64

```
py", line 555, in recreate
    _, user_packages = self.installed_packages()
  File "C:\Users\IEUser\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 925, in installed_packages
    for package, version in self.pip.installed():
  File "C:\Users\IEUser\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 331, in installed
    lines = self.list().splitlines()
  File "C:\Users\IEUser\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 321, in list
    return self.run("list")
  File "C:\Users\IEUser\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 244, in run
    result = self.process.run_blocking(
  File "C:\Users\IEUser\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 120, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\IEUser\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 148, in wait
    output = self.data()
  File "C:\Users\IEUser\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 194, in data
    return output.decode(sys.stdout.encoding, errors="replace")
AttributeError: 'NoneType' object has no attribute 'encoding'

```

carlosperate, 2021-07-02:
The interesting part from the log:
```
2021-07-02 12:39:09,886 - mu.virtual_environment:919(installed_packages) INFO: Discovering installed third party modules in venv.
2021-07-02 12:39:09,886 - mu.virtual_environment:112(run_blocking) INFO: About to run blocking C:\Users\IEUser\AppData\Local\python\mu\mu_venv-38-20210408-215838\scripts\pip.exe with args ['list', '--disable-pip-version-check'] and envvars {}
```

Crashed at:
https://github.com/mu-editor/mu/blob/05ca9e3c23707572aec7f9fc910be9901652d38d/mu/virtual_environment.py#L106-L120
https://github.com/mu-editor/mu/blob/05ca9e3c23707572aec7f9fc910be9901652d38d/mu/virtual_environment.py#L138-L148
https://github.com/mu-editor/mu/blob/05ca9e3c23707572aec7f9fc910be9901652d38d/mu/virtual_environment.py#L191-L194

carlosperate, 2021-07-02:
If I run the embedded Python on its own the encoding it set:
```
C:\Users\IEUser>C:\Users\IEUser\AppData\Local\python\mu\mu_venv-38-20210408-215838\scripts\python.exe
Python 3.8.5 (default, Aug 30 2020, 16:01:16) [MSC v.1927 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.stdout
<_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
>>> sys.stdout.encoding
'utf-8'
```

It's possibly that by setting the env to `{}` we are affecting the stdout encoding?

```
C:\Users\IEUser>SET
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\IEUser\AppData\Roaming
ChocolateyInstall=C:\ProgramData\chocolatey
ChocolateyLastPathUpdate=131976065579984350
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=MSEDGEWIN10
ComSpec=C:\Windows\system32\cmd.exe
DriverData=C:\Windows\System32\Drivers\DriverData
FPS_BROWSER_APP_PROFILE_STRING=Internet Explorer
FPS_BROWSER_USER_PROFILE_STRING=Default
HOMEDRIVE=C:
HOMEPATH=\Users\IEUser
LOCALAPPDATA=C:\Users\IEUser\AppData\Local
LOGONSERVER=\\MSEDGEWIN10
NUMBER_OF_PROCESSORS=2
OneDrive=C:\Users\IEUser\OneDrive
OS=Windows_NT
Path=C:\Program Files (x86)\Parallels\Parallels Tools\Applications;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\ProgramData\chocolatey\bin;C:\Program Files\Puppet Labs\Puppet\bin;C:\Users\IEUser\AppData\Local\Microsoft\WindowsApps;
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 142 Stepping 10, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=8e0a
ProgramData=C:\ProgramData
ProgramFiles=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROMPT=$P$G
PSModulePath=C:\Program Files\WindowsPowerShell\Modules;C:\Windows\system32\WindowsPowerShell\v1.0\Modules
PUBLIC=C:\Users\Public
SESSIONNAME=Console
SystemDrive=C:
SystemRoot=C:\Windows
TEMP=C:\Users\IEUser\AppData\Local\Temp
TMP=C:\Users\IEUser\AppData\Local\Temp
USERDOMAIN=MSEDGEWIN10
USERDOMAIN_ROAMINGPROFILE=MSEDGEWIN10
USERNAME=IEUser
USERPROFILE=C:\Users\IEUser
windir=C:\Windows
```

carlosperate, 2021-07-02:
> It's possibly that by setting the env to `{}` we are affecting the stdout encoding?

No, setting the env var is an additive process:

https://github.com/mu-editor/mu/blob/05ca9e3c23707572aec7f9fc910be9901652d38d/mu/virtual_environment.py#L97-L104

In that case it might be a difference between running from command line vs GUI.

carlosperate, 2021-07-02:
Okay so:

```
>C :\Users\IEUser\AppData\Local\python\mu\mu_venv-38-20210408-215838\scripts\python.exe -c "import sys; a=open('a.txt','w'); a.write(sys.stdout.encoding)"
```
And `a.txt` contains: `utf-8`

```
> C:\Users\IEUser\AppData\Local\python\mu\mu_venv-38-20210408-215838\scripts\pythonw.exe -c "import sys; w=open('w.txt','w'); w.write(sys.stdout.encoding)"
```

And the `w.txt` file is empty. I assume an exception is thrown but because nothing is output to the terminal I don't get to see it. 

Writing `str(sys.stdout)` to the file and we get `None`.

**So it looks like in GUI mode with pythonw.exe we don't get sys.stdout.**

ntoll, 2021-07-02:
Wonderful detective work Sherlock Carlos. :detective:

carlosperate, 2021-07-02:
I've added a simple default to UFT-8 if Mu doesn't have access to `sys.stdout` and `os.device_encoding(0)` due to `pythonw.exe`: https://github.com/mu-editor/mu/commit/b4626fc1a97da240a1307e45cbc31ba4eb63ce0f
It's far from perfect by works on my Windows 10 VM, so that will unblock us for the next beta release.

@tjguk whenever you might have time again (don't worry about this all in the meantime), let me know if you have any thoughts about doing this differently.

carlosperate, 2021-08-11:
Closing as fixed in beta.5

closed 2021-08-11 by carlosperate