mu-editor/mu #1679: Crash Report

riturajs0896, 2021-07-02:
[mu.log](https://github.com/mu-editor/mu/files/6756028/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_IN
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Lenovo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 283, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Lenovo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\Lenovo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Lenovo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "C:\Users\Lenovo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 626, in create
    self.register_baseline_packages()
  File "C:\Users\Lenovo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 751, in register_baseline_packages
    packages = list(self.pip.installed())
  File "C:\Users\Lenovo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 289, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages

```

carlosperate, 2021-07-05:
Thank you for the report and the log @riturajs0896!

We've just released a new beta version of Mu, would you mind installing this version and trying again?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-08-12:
Hi @riturajs0896,

Were you able to try again with beta.5?
https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-09-21:
@riturajs0896 have you been able to test this with beta.5? There has been some changes in this area that we think could have helped with this issue: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.5

carlosperate, 2021-12-16:
Hi @riturajs0896, we found that other crash reports with similar problems were resolved by disabling the antivirus when launching Mu. If you have an antivirus running, could you try disabling it? If that works would you mind sharing what Antivirus software is it? It'd be really useful to know so that we can try to resolve this issue.

carlosperate, 2022-05-31:
Closing due to lack of activity.