mu-editor/mu #1680: Crash Report

madhwaraj-github, 2021-07-03:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_IN
Platform: Linux 5.8.0-59-generic #66~20.04.1-Ubuntu SMP Thu Jun 17 11:14:10 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 730, in install_baseline_packages
    wheels.download(interpreter=self.interpreter, logger=logger)
  File "/usr/local/lib/python3.8/dist-packages/mu/wheels/__init__.py", line 137, in download
    raise WheelsDownloadError(
mu.wheels.WheelsDownloadError: Pip was unable to download pgzero>=1.2.1

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 161, in run
    raise ex
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 494, in ensure_and_create
    self.create()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 625, in create
    self.install_baseline_packages()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 732, in install_baseline_packages
    raise VirtualEnvironmentCreateError(exc.message)
mu.virtual_environment.VirtualEnvironmentCreateError: Pip was unable to download pgzero>=1.2.1

```
[mu.log](https://github.com/mu-editor/mu/files/6757953/mu.log)

carlosperate, 2021-07-03:
Thanks for the report and the logs @madhwaraj-github!

Did you install Mu using pip? If so, did you use sudo to install it?

madhwaraj-github, 2021-07-04:
Yes.
sudo pip3 install mu-editor==1.1.0b4

carlosperate, 2021-07-04:
Does Mu run if you use sudo?

If it does, could you try uninstalling mu (you might need sudo) and then trying to installing without sudo? By default your OS should probably installing than in "user mode" and should work without it.

madhwaraj-github, 2021-07-04:
Yes, it worked when used with sudo.
I subsequently tried:

sudo pip3 uninstall mu-editor
pip3 install mu-editor

install errored out, and the errors are captured in the attached file.

pip3 install mu-editor
Collecting mu-editor
  Downloading mu_editor-1.0.3-py3-none-any.whl (2.0 MB)
     |████████████████████████████████| 2.0 MB 2.1 MB/s 
Collecting qscintilla==2.11.4
  Downloading QScintilla-2.11.4-1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.9 MB)
     |████████████████████████████████| 1.9 MB 7.2 MB/s 
Collecting matplotlib==2.2.2
  Downloading matplotlib-2.2.2.tar.gz (37.3 MB)
     |████████████████████████████████| 37.3 MB 6.1 MB/s 
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-k4atakef/matplotlib/setup.py'"'"'; __file__='"'"'/tmp/pip-install-k4atakef/matplotlib/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-k4atakef/matplotlib/pip-egg-info
         cwd: /tmp/pip-install-k4atakef/matplotlib/
    Complete output (63 lines):
    ============================================================================
    Edit setup.cfg to change the build options
    
    BUILDING MATPLOTLIB
                matplotlib: yes [2.2.2]
                    python: yes [3.8.10 (default, Jun  2 2021, 10:49:15)  [GCC
                            9.4.0]]
                  platform: yes [linux]
    
    REQUIRED DEPENDENCIES AND EXTENSIONS
                     numpy: yes [version 1.17.4]
          install_requires: yes [handled by setuptools]
                    libagg: yes [pkg-config information for 'libagg' could not
                            be found. Using local copy.]
                  freetype: no  [The C/C++ header for freetype2 (ft2build.h)
                            could not be found.  You may need to install the
                            development package.]
                       png: no  [pkg-config information for 'libpng' could not
                            be found.]
                     qhull: yes [pkg-config information for 'libqhull' could not
                            be found. Using local copy.]
    
    OPTIONAL SUBPACKAGES
               sample_data: yes [installing]
                  toolkits: yes [installing]
                     tests: no  [skipping due to configuration]
            toolkits_tests: no  [skipping due to configuration]
    
    OPTIONAL BACKEND EXTENSIONS
                    macosx: no  [Mac OS-X only]
                    qt5agg: yes [installing, Qt: 5.13.2, PyQt: 5.13.2; PySide2
                            not found]
                    qt4agg: no  [PySide not found; PyQt4 not found]
                   gtk3agg: yes [installing, version 3.20.24]
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-k4atakef/matplotlib/setup.py", line 197, in <module>
        msg = pkg.install_help_msg()
      File "/tmp/pip-install-k4atakef/matplotlib/setupext.py", line 592, in install_help_msg
        release = platform.linux_distribution()[0].lower()
    AttributeError: module 'platform' has no attribute 'linux_distribution'
                 gtk3cairo: yes [installing, version 3.20.24]
                    gtkagg: no  [Requires pygtk]
                     tkagg: yes [installing; run-time loading from Python Tcl /
                            Tk]
                     wxagg: no  [requires wxPython]
                       gtk: no  [Requires pygtk]
                       agg: yes [installing]
                     cairo: yes [installing, pycairo version 1.16.2]
                 windowing: no  [Microsoft Windows only]
    
    OPTIONAL LATEX DEPENDENCIES
                    dvipng: no
               ghostscript: yes [version 9.50]
                     latex: no
                   pdftops: yes [version 0.86.1]
    
    OPTIONAL PACKAGE DATA
                      dlls: no  [skipping due to configuration]
    
    ============================================================================
                            * The following required packages can not be built:
                            * freetype, png
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

carlosperate, 2021-07-04:
It looks like without sudo it's trying to install matplotlib from source. Could you try to run these commands to see if they print the same?

```
sudo pip3 --version
```

```
pip3 --version
```

madhwaraj-github, 2021-07-05:
***@***.***:~$ sudo pip3 --version
[sudo] password for raj:
pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)
***@***.***:~$ pip3 --version
pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)

carlosperate, 2021-07-05:
That's a bit odd, if it's the same Python and pip version how come it installed a the `matplotlib` wheel with sudo, but tried to build it the package from source without it? 🤔 

Could you try creating a virtual environment and installing Mu inside that?
Alternatively I really like pipx for this kind of Python application, as it automatically creates a virtual environment and add the executable to path: https://pypa.github.io/pipx/

madhwaraj-github, 2021-07-05:
***@***.***:~$ pipx install mu-editor
Fatal error from pip prevented installation. Full pip output in file:
    /home/raj/.local/pipx/logs/cmd_2021-07-05_16.07.55_pip_errors.log

pip seemed to fail to build package:
    mu-editor

Some possibly relevant errors from pip install:
    numpy: yes [not found. pip may install it below.]
    could not be found.  You may need to install the
    qt5agg: no  [PySide2 not found; PyQt5 not found]
    qt4agg: no  [PySide not found; PyQt4 not found]
    AttributeError: module 'platform' has no attribute 'linux_distribution'
    cairo: no  [cairocffi or pycairo not found]
    ERROR: Cannot install mu-editor==1.0.0, mu-editor==1.0.1,
mu-editor==1.0.2 and mu-editor==1.0.3 because these package versions have
conflicting dependencies.
    ERROR: ResolutionImpossible: for help visit
https://pip.pypa.io/en/latest/user_guide/#fixing-conflicting-dependencies

Error installing mu-editor.

carlosperate, 2021-07-05:
Sorry, I should have clarified you need to specify the version:

```
pipx install mu-editor==1.1.0b5
```

madhwaraj-github, 2021-07-05:
Yes, that worked. Thank you!

carlosperate, 2021-07-05:
That's awesome, thanks for the report and all the testing!

closed 2021-07-05 by carlosperate