mu-editor/mu #1681: Crash Report

grodriguez0149, 2021-07-03:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: es_AR
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Edgardo Roatta\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Edgardo Roatta\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Edgardo Roatta\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\Edgardo Roatta\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\Edgardo Roatta\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 344, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\Edgardo Roatta\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 798, in setup
    os.makedirs(wd)
  File "C:\Users\Edgardo Roatta\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] No se puede crear un archivo que ya existe: 'C:\\Users\\Edgardo Roatta\\mu_code'

```

carlosperate, 2021-07-03:
Thanks for the report @grodriguez0149!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Edgardo Roatta\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-08-17:
Hola @grodriguez0149 

Gracias por enviar el Crash Report. Podrias tambien enviar los logs de la applicacion? No ayudarian bastante en diagnosticar el problem. El archivo deberia estar en `C:\Users\Edgardo Roatta\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-09-21:
I'll close this due to lack of activity, it looks like the race condition already tracked in #1755.

@grodriguez0149 if you are still experiencing this problem please reopen this issue, thanks!
@grodriguez0149 si todavia tienes este problema, por favor vuelve a abrir este GitHub issue y podremos investigarlo, gracias!

closed 2021-09-21 by carlosperate