mu-editor/mu #1682: Fails to Flash

mitch46syd, 2021-07-03:
Hi

just acquired a Micro.bit kit
running it on a MacBook Pro mid 2015
[image]

Osx 10.15.7 (19H1217) Catalina
MU 1.1.0.beta.4 
Cannot Flash
would it be conflicting with my existing Python ? - 3.7.4
[MU Log 03 Jul 2021_2205.txt](https://github.com/mu-editor/mu/files/6758368/MU.Log.03.Jul.2021_2205.txt)

tia

carlosperate, 2021-07-03:
Thanks for the report @mitch46syd!

The Python version in your computer should not affect Mu, I think this might be a permissions issue. Could you follow these steps to see if it works:

https://github.com/mu-editor/mu/issues/1383#issuecomment-810537546

mitch46syd, 2021-07-04:
Hi Carlos

just what the patient wanted!!

tks for prompt reply

> 
> 
> Thanks for the report @mitch46syd <https://github.com/mitch46syd>!
> 
> The Python version in your computer should not affect Mu, I think this might be a permissions issue. Could you follow these steps to see if it works:
> 
> #1383 (comment) <https://github.com/mu-editor/mu/issues/1383#issuecomment-810537546>

regards

John

CQR Pty Ltd
1005/2 Dind Street, Milsons Point, NSW 2061 Australia 
ABN: 45 001 785 191 
Email:  ***@***.*** • Mob: +61 410 318 131
Skype: jrmitchell

carlosperate, 2021-07-04:
Great, glad it worked!

closed 2021-07-04 by carlosperate