mu-editor/mu #1683: Crash Report while opening Mu micropython

paco218, 2021-07-05:
Version: 1.1.0.beta.5
Locale: es_MX
Platform: Darwin 17.7.0 Darwin Kernel Version 17.7.0: Mon Aug 31 22:11:23 PDT 2020; root:xnu-4570.71.82.6~1/RELEASE_X86_64 x86_64

```
ate
    self.install_baseline_packages()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: debugpy-1.3.0-cp38-cp38-macosx_10_14_x86_64.whl is not a supported wheel on this platform.

```

carlosperate, 2021-07-05:
Thanks for the report @paco218!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `~/Library/Logs/mu`

What version of macOS are you using?

carlosperate, 2021-07-06:
Okay, so based on the Darwin version it looks like you are in 10.13 High Sierra.

I believe the problem is from the recent debug capabilities added to `ipykerner` v6 released less than a week ago on June 30th:
- https://github.com/ipython/ipykernel/pull/597
- https://github.com/ipython/ipykernel/releases/tag/6.0.0

This added the `debugpy` dependency, which is only compatible with 10.14 Mojave upwards.
The fix could be to force the use `ipykerner` v5.5.5, which doesn't have this dependency:
- v5: https://github.com/ipython/ipykernel/blob/5.5.5/setup.py#L87-L93
- v6: https://github.com/ipython/ipykernel/blob/6.0.0/setup.py#L78

carlosperate, 2021-07-06:
@paco218

carlosperate, 2021-07-06:
@paco218 could you download the macOS installer from here and see if it works for you? It's the second artifact called `
mu-editor-macOS-6da71737b3259c0b4a8fcfadda25077318cee2bc`:
https://github.com/mu-editor/mu/actions/runs/1005168883

Because this installer is not signed you'll have to right click on the `Mu Editor.app` icon, and select "Open". I will likely ask you to confirm, so you'll have to accept that as well.

carlosperate, 2021-07-08:
I've tested this in a macOS 10.13 VM and it still fails.

This is what happens:
1. pip first downloads the right version of `ipykerner`
2. Then when downloading `qtconsole` pip always tries to download the latest version of the dependencies available
    - This means that it checks PyPI and it checks the directory set together with `--find-links`, if there is a newer wheel in PyPI it will download it
3. At this point Mu has 2 `ipykerner` wheels, one with version v5.5.5 (good), and one for v6 (bad)
4. ❌ When Mu is opened in macOS High Sierra it first tries to install `ipykerner` v6 and fails the same way

There are a couple of options:
- Use a constrains file: https://pip.pypa.io/en/stable/user_guide/#constraints-files
    - This would be a perfect solution, tested it and works wonderfully
    - ❌ However is only available in pip 20.2+
        - Because we need to maximise compatibility with a system installed virtualenv (like those installed from debian packages), the pip version vendored in most of these would not be new enough :(
- We could add all the `qtconsole` dependencies to `mu/wheels/__init__.py` pinned to our requirements and then download `qtconsole` with the flag `--no-deps`
    - This works fine for downloading the wheels we need and zipping them up
    - ❌ However, on startup Mu tries to verify the venv installed modules. At this point it fails while verying `pyqt` because no Qt python package is found, so it returns an error code, and Mu fails to start up
        - This lack of PyQt5/PySide packages in the venv perfectly fine for our usage, we install `qtconsole` to have exactly the same ipython packages installed in Mu and the user venv, so we don't need Qt in there as well
- In `mu/wheels/__init__.py` we could install only `ipykerner` instead of `qtconsole`
    - 🤷‍♂️This could potentially work, but I would then prefer to pin the same version of `ipykerner` in setup.py as well
- We could update the venv creation to install the iPython kernel using the Mu python interpreter and targetting the user venv
    - Right now the iPython dependencies are installed in the user venv, and then the venv is activated to install `ipykerner` in itself
    - This is more work and we haven't really tested this before, so it could have the potential to have unexpected bugs in some user environments
    - 🤷‍♂️ It has the potential to work, reduce the total package size and reduce the startup time

carlosperate, 2021-07-08:
Okay so option 4 is not possible:
> We could update the venv creation to install the iPython kernel using the Mu python interpreter and targetting the user venv 

We are actually installing the kernel from Mu's interpreter, not the user venv. We then launch configure the venv interpreter to be used with the kernel here: 
https://github.com/mu-editor/mu/blob/6359ac12189065118ab9467a5847151ce7894dbb/mu/modes/python3.py#L52

So the venv must have the `ipykerner` package installed for this work.

That leaves us option 3:
>  In `mu/wheels/__init__.py` we could install only `ipykerner` instead of `qtconsole`

So I've raised PR https://github.com/mu-editor/mu/pull/1692 and will test in the macOS 10.13 High Sierra again as soon as the CI spits out an installer.

carlosperate, 2021-07-09:
I've tested the installer from PR #1692 and it works 🎉 

@paco218 this will be fixed in the next release, as a workaround if you like you can install this (unsigned) version of Mu. It's the second artefact called `mu-editor-macOS-dc5e22f35095684df25f16341c9ec4a473d1da35 `: https://github.com/mu-editor/mu/actions/runs/1013306335

Because this is a testing installer it not signed, so to run it you'll have to right click on the `Mu Editor.app` icon, and select "Open". I will ask you to confirm, so you'll have to accept that as well.

carlosperate, 2021-10-06:
This has been fix in beta.6: https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.6 🎉

closed 2021-10-06 by carlosperate