mu-editor/mu #1684: Crash report - failed to import esptool

emilia-szymanska, 2021-07-06:
Hi! I did the ```pip3 install mu-editor``` and wanted to run the ```mu-editor``` command. The following issue appeared:

```
Traceback (most recent call last):
  File "/home/emilia/.local/lib/python3.8/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/emilia/.local/lib/python3.8/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/emilia/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "/home/emilia/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "/home/emilia/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 602, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: esptool

```
I tried to install esptool manually, it didn't help.

Version: 1.1.0.beta.4
Locale: en_NG
Platform: Linux 5.8.0-59-generic #66~20.04.1-Ubuntu SMP Thu Jun 17 11:14:10 UTC 2021 x86_64

The log file is as follows: [mu.log](https://github.com/mu-editor/mu/files/6769400/mu.log)

emilia-szymanska, 2021-07-06:
I uninstalled the version of mu-editor I had and installed again. It seems to be working, although I had some errors while installing:
```
ERROR: flake8 3.9.2 has requirement pycodestyle<2.8.0,>=2.7.0, but you'll have pycodestyle 2.4.0 which is incompatible.
ERROR: flake8 3.9.2 has requirement pyflakes<2.4.0,>=2.3.0, but you'll have pyflakes 2.0.0 which is incompatible.
```
Apart from these errors everything seems OK, mu-editor finally launched!

carlosperate, 2021-07-06:
Thanks for the report @emilia-szymanska!

There is a new beta.5 out, is it possible that was the version that worked?
```
pip3 install mu-editor==1.1.0b5
``` 

Apart from that it's highly recommended to use a virtual environment to install Mu. I personally really like pipx to install this type of Python applications, it manages a venv for each installation, adding the executables to path, etc: https://pypa.github.io/pipx/

carlosperate, 2021-08-10:
Hi @emilia-szymanska, just wondering if you had a chance to try beta.5?

emilia-szymanska, 2021-08-10:
Hi @carlosperate, sorry for not responding! I tried downloading the newest version and it worked!

carlosperate, 2021-08-10:
No worries, I'm glad beta 5 worked, thanks for the update!

closed 2021-08-10 by carlosperate