mu-editor/mu #1687: venv: Downloading wheels does not show progress in the splash screen

carlosperate, 2021-07-07:
### What were you trying to do?

Running Mu for the first time without zipped wheels

tjguk, 2021-07-09:
Yes; I spotted this when developing but couldn't quickly see any reason (in principle the "splash screen" logger is passed up the chain). So I left it there for now.

carlosperate, 2021-08-04:
It was pretty much all setup to be able to do this already, just had to pass the logger as an argument to `wheels.download()`: https://github.com/mu-editor/mu/pull/1735

closed 2021-09-20 by ntoll