mu-editor/mu #1690: Crash Report

ChiangFamily, 2021-07-08:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: zh_TW
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\dawei\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 161, in run
    raise ex
  File "C:\Users\dawei\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\dawei\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\dawei\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 767, in create
    self.create_venv()
  File "C:\Users\dawei\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 798, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\dawei\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\dawei\AppData\Local\python\mu\mu_venv-38-20210708-214825
AttributeError: module 'virtualenv.create.via_global_ref.builtin.cpython.mac_os' has no attribute 'CPython3macOsFramework'

```

carlosperate, 2021-07-08:
Hi @ChiangFamily, thanks for the report!

It looks like that kind of error usually happens with a bad installation: https://github.com/pypa/virtualenv/issues/1740

Could you uninstall Mu, download the beta.5 installer again (make sure you download it again, not just use an installer you might have used before), install it, and see if the problem persists?

carlosperate, 2021-08-11:
Hi @ChiangFamily,

I was wondering if you had the change to uninstall and install again the latest beta? https://github.com/mu-editor/mu/releases/

carlosperate, 2021-08-11:
Hi @ChiangFamily,

Thanks for trying again!
Could you list the steps here you took to uninstall and reinstall Mu? Thanks!

carlosperate, 2021-09-24:
Hi @ChiangFamily

Would you be able to run these command in a "command prompt"? It will help us identify the issue:

```
"C:\Users\dawei\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.path)"
```
```
"C:\Users\dawei\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
```

ChiangFamily, 2021-09-24:
Now, My computer can run Mu without any problems.
The version i installed is Mu-Editor-Win64-1.1.0b5.msi.

carlosperate, 2021-09-26:
That's great, thanks for the update, I'll close this as resolve then. Thanks for all the testing!

closed 2021-09-26 by carlosperate