mu-editor/mu #1691: Crash Report

adrpetwad, 2021-07-08:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.4
Locale: en_US
Platform: Linux 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l

```
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/pi/.local/lib/python3.7/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/pi/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 495, in ensure_and_create
    self.ensure()
  File "/home/pi/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 524, in ensure
    self.ensure_key_modules()
  File "/home/pi/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 603, in ensure_key_modules
    "Failed to import: %s" % module
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: pgzero

```

carlosperate, 2021-07-08:
Hi @adrpetwad,

Thanks for the report! There is a new beta 5 release, could you install that version and try again?

carlosperate, 2021-08-04:
Hi @adrpetwad, I was just wondering if you had a chance to install the beta 5 and try again?

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate