mu-editor/mu #1701: not responding after copy + paste

u-nico, 2021-07-13:
### What were you trying to do?

copy a number from https://www.farbtabelle.at/farben-umrechnen/ to mu

### What steps did you take to trigger the issue?

i went to https://www.farbtabelle.at/farben-umrechnen/ -> typed in hex -> went to decimal -> copied decimal -> went to mu -> strg + v

### What did you expect to happen?

the number placed

### What actually happened?

if strg-v once: i couldn't do anything for some seconds
if multiple times: mu hang up

### Operating System Version

Ubuntu 21.04

### Mu Version

Mu 1.0.3

### Other Info

other os info:
64-bit
GNOME-Version: 3.38.5
Window-manager: Wayland

hardware info:
Intel i5-6200U

### Editor Log

sry, but upload didn't work and comment is too long (if you need it: send me a message)

abixadamj, 2021-07-14:
Please add screenshot how mu looks, when you paste number to it...

ntoll, 2021-07-14:
I cannot reproduce this on Ubuntu 20.04. Perhaps this is a problem local to your system..? Here's what I see:

[image]

u-nico, 2021-07-14:
oh, I tried to do it with Text-Editor and then nothing happened, so you're probably right and it's a problem with my system. Next time, I'll try it with other Text-Editors first, sry.

closed 2021-07-14 by u-nico