mu-editor/mu #1702: Crash Report

yongzhang2, 2021-07-15:
[07152021 report.docx](https://github.com/mu-editor/mu/files/6826379/07152021.report.docx)
[07152021 report.docx](https://github.com/mu-editor/mu/files/6826381/07152021.report.docx)
[**REMEMBER]([07152021 report.docx](https://github.com/mu-editor/mu/files/6826376/07152021.report.docx)) TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\yong\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\yong\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\yong\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\yong\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\yong\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 351, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\yong\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 798, in setup
    os.makedirs(wd)
  File "C:\Users\yong\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\yong\\mu_code'

```

carlosperate, 2021-08-17:
Hi @yongzhang2,

Thanks for the report and attaching the logs!

Looking at the logs all entries are duplicated, which makes me believe that there were perhaps 2 instances of Mu running in parallel and racing each other at creating these directories.

If you try to run Mu again, and ensure it is only running once, do you still get this problem?

carlosperate, 2021-09-21:
Hi @yongzhang2, I was wondering if you were still experiencing this issue?

carlosperate, 2021-12-17:
Hi @yongzhang2, are you still experiencing this problem with Mu?

carlosperate, 2022-01-19:
As this problem is captured in #1755, I'll close this issue to keep things tidy.
Thanks for the report!

closed 2022-01-19 by carlosperate