mu-editor/mu #1705: Crash Report

Lichen-the-green, 2021-07-16:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
d_wheels_filepath)
  File "C:\Users\********\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\********\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\*********\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\********\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\********\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 101:
Unable to create process using 'C:\Users\********\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\******\AppData\Local\python\mu\mu_venv-38-20210716-125806\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\ABCDEF~1\AppData\Local\Temp\tmprdcthh3_\backcall-0.2.0-py2.py3-none-any.whl'

```

carlosperate, 2021-08-05:
Thanks for the report @Lichen-the-green!

Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `%LOCALAPPDATA%\python\mu\Logs\mu.log`

Also, is it possible you were running an antivirus while starting up Mu? If so, would you be able to disable it and try again?

carlosperate, 2021-08-06:
Oh, and does your Windows username has a space?

closed 2022-05-31 by carlosperate