mu-editor/mu #1706: Crash Report

IAliKhanJaun, 2021-07-17:
**REMEMBER TO ATTACH YOUR LOG FILE**
C:\Users\Iman\AppData\Local\python\mu\Logs\mu.log
Version: 1.1.0.beta.5
Locale: en_CH
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Iman\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Iman\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Iman\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\Iman\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\Iman\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 351, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\Iman\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 798, in setup
    os.makedirs(wd)
  File "C:\Users\Iman\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\Iman\\mu_code'

```

carlosperate, 2021-08-10:
Thanks for the report @IAliKhanJaun!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging.

The path to file was added to the issue, but not the file itself. It should be located in: `C:\Users\Iman\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-09-21:
I'll close this due to lack of activity, it looks like the race condition already tracked in #1755.

@IAliKhanJaun if you are still experiencing this problem please reopen this issue, thanks!