mu-editor/mu #1707: Dark Mode should support f"{x}:{y}"

rsbohn12, 2021-07-17:
### What were you trying to do?

Use an f-string

### What steps did you take to trigger the issue?

```python
x=20
y=30
print(f"{x}:{y}")
```

### What did you expect to happen?

f-strings should look like other strings

### What actually happened?

The f-string is purple on white. The variables (x,y) are gray on black.

### Operating System Version

Windows 10

### Mu Version

1.1.0-beta-5

FoamyGuy, 2021-07-17:
Here is a screenshot of how the fstrings appear in dark mode:
[image]

ntoll, 2021-07-19:
Hmm.... nice catch.

commit 7475770c (2021-12-03): Merge pull request #1900 from devdanzin/enable_fstrings_dark_mode

Enable f-strings in dark modes. Fixes #1707.
closed 2021-12-03 by carlosperate