mu-editor/mu #1708: Crash Report

chengesbaugh, 2021-07-18:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Darwin 17.7.0 Darwin Kernel Version 17.7.0: Thu Jan 23 07:05:23 PST 2020; root:xnu-4570.71.69~1/RELEASE_X86_64 x86_64

```
_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/private/var/folders/1q/682yx6_d7_96b_jk0n332q_00000gn/T/AppTranslocation/4A29E9CA-EA26-40F1-AC9E-A9F17E30378C/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/private/var/folders/1q/682yx6_d7_96b_jk0n332q_00000gn/T/AppTranslocation/4A29E9CA-EA26-40F1-AC9E-A9F17E30378C/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/private/var/folders/1q/682yx6_d7_96b_jk0n332q_00000gn/T/AppTranslocation/4A29E9CA-EA26-40F1-AC9E-A9F17E30378C/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/private/var/folders/1q/682yx6_d7_96b_jk0n332q_00000gn/T/AppTranslocation/4A29E9CA-EA26-40F1-AC9E-A9F17E30378C/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: debugpy-1.3.0-cp38-cp38-macosx_10_14_x86_64.whl is not a supported wheel on this platform.

```

chengesbaugh, 2021-07-18:
I am learning python for the first time. This is my error upon opening Mu for the very first time. I am working on a Mac; do I need to install an older version of Mu, because I have not updated my Mac?

carlosperate, 2021-07-19:
Hi @chengesbaugh, thanks for the report!

Are you in macOS 10.13 High Sierra? If so, could you try the installer from https://github.com/mu-editor/mu/issues/1683#issuecomment-877058288 ?
Thanks!

chengesbaugh, 2021-07-24:
It worked! Thank you :)

carlosperate, 2021-08-05:
Thanks for the update, glad it worked!
I'll close this issue as resolved since the next Mu beta release will include this fix.
Thanks for the testing!

closed 2021-08-05 by carlosperate