mu-editor/mu #1710: minus sign shown as reyed out.

mpechner, 2021-07-18:
### What were you trying to do?

Read a json file and display the data.

### What steps did you take to trigger the issue?

programming a adafruit magtag. Snippet. On my retina it showed, on my dell the minus on lat/lon data was so faint I did not see it.  I have no such issue with pycharm, cat, less, vim.

eq_url = "http://wedding.mikey.com/fmt_c.json"
magtag.url = eq_url
fetched_data = magtag.fetch()
print("FETCHED", fetched_data)

### What did you expect to happen?

To see that '-' minus clearly.

### What actually happened?

On the dell not visible on retina visible.

Even tried dark mode.

Screen capture external dell monitor
[image]

Screen capture retina display.  2018 MBP
[image]

### Operating System Version

OSC 11.0.1

### Mu Version

1.1.0beta5

ntoll, 2021-07-19:
Hi @mpechner 

Wow, that's a weird one. Mu is just using a standard font in a standard way and is certainly doing nothing more than that.

That you don't get the problem on a different monitor, and that I can't reproduce this (I checked), suggests this might be something about your local setup (but what that might be, I'm not sure.... see the "wow, that's a weird one" comment above).

I'm not sure what to suggest.

N.

carlosperate, 2021-07-19:
Sounds similar to this?

ntoll, 2021-07-19:
Ahh.... great memory there @carlosperate - hmm... so it appears to be an upstream PyQt / Qt5 thing... :-/