mu-editor/mu #1712: Crash Report

jsnyder1, 2021-07-19:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
n_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 351, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 829, in setup
    shutil.copytree(path("templates", "web/"), template_path)
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 554, in copytree
    return _copytree(entries=entries, src=src, dst=dst, symlinks=symlinks,
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 455, in _copytree
    os.makedirs(dst, exist_ok=dirs_exist_ok)
  File "C:\Users\Admin\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\Admin\\mu_code\\templates'

```
C:\Users\Admin\AppData\Local\python\mu\Logs\mu.log:

2021-07-19 16:35:44,146 - root:271(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.5
2021-07-19 16:35:44,151 - root:272(run) INFO: uname_result(system='Windows', node='AcerS3', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 42 Stepping 7, GenuineIntel')
2021-07-19 16:35:44,153 - root:273(run) INFO: Platform: Windows-10-10.0.19041-SP0
2021-07-19 16:35:44,153 - root:274(run) INFO: Python path: ['C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-07-19 16:35:44,178 - root:275(run) INFO: Language code: en_US
2021-07-19 16:35:44,428 - root:271(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.5
2021-07-19 16:35:44,428 - root:272(run) INFO: uname_result(system='Windows', node='AcerS3', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 42 Stepping 7, GenuineIntel')
2021-07-19 16:35:44,428 - root:273(run) INFO: Platform: Windows-10-10.0.19041-SP0
2021-07-19 16:35:44,428 - root:274(run) INFO: Python path: ['C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-07-19 16:35:44,428 - root:275(run) INFO: Language code: en_US
2021-07-19 16:35:47,031 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2021-07-19 16:35:47,063 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2021-07-19 16:35:53,821 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-07-19 16:35:53,821 - mu.virtual_environment:777(create_venv) INFO: Creating virtualenv: C:\Users\Admin\AppData\Local\python\mu\mu_venv-38-20210719-163436
2021-07-19 16:35:53,821 - mu.virtual_environment:777(create_venv) INFO: Creating virtualenv: C:\Users\Admin\AppData\Local\python\mu\mu_venv-38-20210719-163503
2021-07-19 16:35:54,071 - mu.virtual_environment:778(create_venv) INFO: Virtualenv name: mu_venv-38-20210719-163436
2021-07-19 16:35:54,071 - mu.virtual_environment:778(create_venv) INFO: Virtualenv name: mu_venv-38-20210719-163503
2021-07-19 16:35:54,086 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-m', 'virtualenv', '-p', 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\Admin\\AppData\\Local\\python\\mu\\mu_venv-38-20210719-163436') with kwargs {'env': {'ALLUSERSPROFILE': 'C:\\ProgramData', 'APPDATA': 'C:\\Users\\Admin\\AppData\\Roaming', 'ASL.LOG': 'Destination=file', 'COMMONPROGRAMFILES': 'C:\\Program Files\\Common Files', 'COMMONPROGRAMFILES(X86)': 'C:\\Program Files (x86)\\Common Files', 'COMMONPROGRAMW6432': 'C:\\Program Files\\Common Files', 'COMPUTERNAME': 'ACERS3', 'COMSPEC': 'C:\\WINDOWS\\system32\\cmd.exe', 'DRIVERDATA': 'C:\\Windows\\System32\\Drivers\\DriverData', 'FPS_BROWSER_APP_PROFILE_STRING': 'Internet Explorer', 'FPS_BROWSER_USER_PROFILE_STRING': 'Default', 'FP_NO_HOST_CHECK': 'NO', 'GTK_BASEPATH': 'C:\\Program Files (x86)\\GtkSharp\\2.12\\', 'HOMEDRIVE': 'C:', 'HOMEPATH': '\\Users\\Admin', 'KICAD_PTEMPLATES': 'C:\\Program Files (x86)\\KiCad\\share\\kicad\\template', 'KISYS3DMOD': 'C:\\Program Files (x86)\\KiCad\\share\\kicad\\modules\\packages3d', 'KISYSMOD': 'C:\\Program Files (x86)\\KiCad\\share\\kicad\\modules', 'LOCALAPPDATA': 'C:\\Users\\Admin\\AppData\\Local', 'LOGONSERVER': '\\\\ACERS3', 'NUMBER_OF_PROCESSORS': '4', 'ONEDRIVE': 'C:\\Users\\Admin\\OneDrive', 'OS': 'Windows_NT', 'PATH': 'C:\\Users\\Admin\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\PyQt5\\Qt\\bin;C:\\ProgramData\\Oracle\\Java\\javapath;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\Program Files (x86)\\QuickTime\\QTSystem\\;C:\\Program Files (x86)\\GtkSharp\\2.12\\bin;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Users\\Admin\\AppData\\Local\\Mu\\bin;C:\\Users\\Admin\\AppData\\Local\\Microsoft\\WindowsApps;', 'PATHEXT': '.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC', 'PROCESSOR_ARCHITECTURE': 'AMD64', 'PROCESSOR_IDENTIFIER': 'Intel64 Family 6 Model 42 Stepping 7, GenuineIntel', 'PROCESSOR_LEVEL': '6', 'PROCESSOR_REVISION': '2a07', 'PROGRAMDATA': 'C:\\ProgramData', 'PROGRAMFILES': 'C:\\Program Files', 'PROGRAMFILES(X86)': 'C:\\Program Files (x86)', 'PROGRAMW6432': 'C:\\Program Files', 'PSMODULEPATH': 'C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules\\', 'PUBLIC': 'C:\\Users\\Public', 'SESSIONNAME': 'Console', 'SYSTEMDRIVE': 'C:', 'SYSTEMROOT': 'C:\\WINDOWS', 'TEMP': 'C:\\Users\\Admin\\AppData\\Local\\Temp', 'TMP': 'C:\\Users\\Admin\\AppData\\Local\\Temp', 'USERDOMAIN': 'AcerS3', 'USERDOMAIN_ROAMINGPROFILE': 'AcerS3', 'USERNAME': 'Admin', 'USERPROFILE': 'C:\\Users\\Admin', 'WINDIR': 'C:\\WINDOWS', '__COMPAT_LAYER': 'DetectorsAppHealth', 'QT_API': 'pyqt5', 'QT_AUTO_SCREEN_SCALE_FACTOR': '1', 'QT_MAC_WANTS_LAYER': '1'}}
[... cut ...]
2021-07-19 16:43:54,964 - mu.modes.base:61(get_default_workspace) INFO: Using workspace C:\Users\Admin\mu_code from settings file
2021-07-19 16:43:54,964 - mu.logic:1523(change_mode) INFO: Workspace directory: C:\Users\Admin\mu_code
2021-07-19 16:43:54,965 - mu.logic:867(restore_session) DEBUG: <SessionSettings from C:\Users\Admin\AppData\Local\python\mu\session.json>
2021-07-19 16:43:55,058 - mu.logic:1470(select_mode) INFO: Showing available modes: ['python', 'circuitpython', 'microbit', 'esp', 'web', 'pyboard', 'debugger', 'pygamezero', 'lego', 'pico']

carlosperate, 2021-12-17:
Thanks for the report @jsnyder1, looks like the issue here might have been #1755.

Assuming that's the case Mu probably run correctly the next time you tried to launch it, is that correct? Or are you still experiencing this problem?

carlosperate, 2022-01-19:
As this problem is captured in #1755, I'll close this issue to keep things tidy.
Thanks for the report!

closed 2022-01-19 by carlosperate