mu-editor/mu #1713: Crash Report

zyxq321, 2021-07-21:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_CA
Platform: Darwin 16.7.0 Darwin Kernel Version 16.7.0: Thu Jun 15 17:36:27 PDT 2017; root:xnu-3789.70.16~2/RELEASE_X86_64 x86_64

```
ate
    self.install_baseline_packages()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: debugpy-1.3.0-cp38-cp38-macosx_10_14_x86_64.whl is not a supported wheel on this platform.

```

carlosperate, 2021-07-21:
Hi @zyxq321, thanks for the report!

Are you in macOS 10.12 Sierra? If so, could you try the installer from https://github.com/mu-editor/mu/issues/1683#issuecomment-877058288 ?

Thanks!

zyxq321, 2021-07-21:
It worked, thanks!

Hi @zyxq321<https://github.com/zyxq321>, thanks for the report!

Are you in macOS 10.12 Sierra? If so, could you try the installer from #1683 (comment)<https://github.com/mu-editor/mu/issues/1683#issuecomment-877058288> ?

Thanks!

carlosperate, 2021-08-05:
Thanks for the update, glad it worked!
I'll close this issue as resolved since the next Mu beta release will include this fix.
Thanks for the testing!

closed 2021-08-05 by carlosperate