mu-editor/mu #1716: Crash Report

luzegrace, 2021-07-22:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Darwin 17.7.0 Darwin Kernel Version 17.7.0: Fri Oct  4 23:08:59 PDT 2019; root:xnu-4570.71.57~1/RELEASE_X86_64 x86_64

```
ate
    self.install_baseline_packages()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: debugpy-1.3.0-cp38-cp38-macosx_10_14_x86_64.whl is not a supported wheel on this platform.

```

luzegrace, 2021-07-22:
I tried to find the log report but can't find it anywhere at all in my computer.

carlosperate, 2021-07-23:
Hi @luzegrace, thanks for the report!

Are you in macOS 10.13 High Sierra? If so, could you try the installer from https://github.com/mu-editor/mu/issues/1683#issuecomment-877058288 ?

Thanks!

carlosperate, 2021-08-05:
Some users with the same issue have reported that the installer in https://github.com/mu-editor/mu/issues/1683#issuecomment-877058288 with the fixed worked.

That being the case, I'll close this issue as resolved since the next Mu beta release will include the fix.
Issue https://github.com/mu-editor/mu/issues/1683 will remain open until the release is out, so feel free to add any additional info over there if you are still having this problem.

Thanks for the testing!

closed 2021-08-05 by carlosperate