mu-editor/mu #1722: Crash Report

Husamfara, 2021-07-26:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_MY
Platform: Windows 8.1 6.3.9600 AMD64

```
rtualenv\seed\embed\via_app_data\via_app_data.py", line 57, in _install
    installer.install(creator.interpreter.version_info)
  File "C:\Users\Husamfara\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 42, in install
    self._sync(filename, into)
  File "C:\Users\Husamfara\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\copy.py", line 13, in _sync
    copy(src, dst)
  File "C:\Users\Husamfara\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\util\path\_sync.py", line 53, in copy
    method(norm(src), norm(dest))
  File "C:\Users\Husamfara\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\util\path\_sync.py", line 64, in copytree
    shutil.copy(src_f, dest_f)
  File "C:\Users\Husamfara\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 415, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "C:\Users\Husamfara\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 279, in copyfile
    _copyfileobj_readinto(fsrc, fdst, min(file_size, COPY_BUFSIZE))
  File "C:\Users\Husamfara\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 185, in _copyfileobj_readinto
    n = fsrc_readinto(mv)
OSError: [Errno 9] Bad file descriptor

```

carlosperate, 2021-08-05:
Thanks for the report @Husamfara !
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Husamfara\AppData\Local\python\mu\Logs\mu.log`

Husamfara, 2021-08-28:
> @Husamfara would you be able to attach the Mu logs to this issue? It will really help us diagnose the problem.
>
> The file should be located in: C:\Users\Husamfara\AppData\Local\python\mu\Logs\mu.log
>
> Thanks!

Best Regards
HOSSAM ADDEN ALFARRA

carlosperate, 2021-08-30:
Hi @Husamfara,

Thank you for getting back to us. Unfortunately GitHub doesn't grab the email attachments, so the logs have to be sent via the thread in the GitHub website.
Would you be able to go into this URL and attach the logs to a response? https://github.com/mu-editor/mu/issues/1722

Thank you again!

carlosperate, 2021-09-21:
Hi @Husamfara, would you be able to send us the log via the GitHub website? https://github.com/mu-editor/mu/issues/1722

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate