mu-editor/mu #1723: Crash Report

fusee203, 2021-07-28:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Windows 10 10.0.17763 AMD64

```
m_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\Windows 10\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\Windows 10\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\Windows 10\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\Windows 10\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\Windows 10\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 101:
Unable to create process using 'C:\Users\Windows 10\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Windows 10\AppData\Local\python\mu\mu_venv-38-20210728-134213\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\WINDOW~1\AppData\Local\Temp\tmpl99fdqu7\backcall-0.2.0-py2.py3-none-any.whl'

```

carlosperate, 2021-08-02:
Thanks for the report @fusee203!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Windows 10\AppData\Local\python\mu\Logs\mu.log`

stienman, 2021-09-07:
I'm getting the same error.  Here's my crash log:

[mu.log](https://github.com/mu-editor/mu/files/7123870/mu.log)

The first error in the log is `2021-09-07 14:46:04,466 - mu.virtual_environment:185(wait) ERROR: Unable to create process using 'C:\Users\Adam Davis\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Adam Davis\AppData\Local\python\mu\mu_venv-38-20210907-144542\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\ADAMDA~1\AppData\Local\Temp\tmp6t5topei\backcall-0.2.0-py2.py3-none-any.whl'
2021-09-07 14:46:04,500 - mu.virtual_environment:643(ensure_and_create) ERROR: Process finished but with error code 101:`

It continues and appears to attempt to run the command again, and failing again:

`2021-09-07 14:46:07,594 - mu.virtual_environment:185(wait) ERROR: Unable to create process using 'C:\Users\Adam Davis\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Adam Davis\AppData\Local\python\mu\mu_venv-38-20210907-144604\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\ADAMDA~1\AppData\Local\Temp\tmp1p0x9gp0\backcall-0.2.0-py2.py3-none-any.whl'
2021-09-07 14:46:07,622 - mu.virtual_environment:643(ensure_and_create) ERROR: Process finished but with error code 101:`

stienman, 2021-09-07:
Running Mu as administrator doesn't resolve the issue.