mu-editor/mu #1725: Crash Report

MrCurly2u, 2021-07-29:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Larry\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Larry\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Larry\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\Larry\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\Larry\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 351, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\Larry\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 798, in setup
    os.makedirs(wd)
  File "C:\Users\Larry\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\Larry\\mu_code'

```2021-07-28 17:45:41,520 - root:271(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.5
2021-07-28 17:45:41,520 - root:272(run) INFO: uname_result(system='Windows', node='Larry-PC', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 37 Stepping 5, GenuineIntel')
2021-07-28 17:45:41,520 - root:273(run) INFO: Platform: Windows-10-10.0.19041-SP0
2021-07-28 17:45:41,520 - root:274(run) INFO: Python path: ['C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-07-28 17:45:41,520 - root:275(run) INFO: Language code: en_US
2021-07-28 17:45:41,520 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Larry\AppData\Local\python\mu\settings.json; skipping
2021-07-28 17:45:41,520 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Larry\AppData\Local\python\mu\session.json; skipping
2021-07-28 17:45:44,263 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2021-07-28 17:45:50,020 - root:271(run) INFO: 

-----------------

Starting Mu 1.1.0.beta.5
2021-07-28 17:45:50,020 - root:272(run) INFO: uname_result(system='Windows', node='Larry-PC', release='10', version='10.0.19041', machine='AMD64', processor='Intel64 Family 6 Model 37 Stepping 5, GenuineIntel')
2021-07-28 17:45:50,020 - root:273(run) INFO: Platform: Windows-10-10.0.19041-SP0
2021-07-28 17:45:50,020 - root:274(run) INFO: Python path: ['C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-49.6.0-py3.8.egg', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-20.2.2-py3.8.egg', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-07-28 17:45:50,020 - root:275(run) INFO: Language code: en_US
2021-07-28 17:45:50,020 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Larry\AppData\Local\python\mu\settings.json; skipping
2021-07-28 17:45:50,020 - mu.settings:220(load) WARNING: No settings file found at C:\Users\Larry\AppData\Local\python\mu\session.json; skipping
2021-07-28 17:45:50,138 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2021-07-28 17:45:52,515 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-07-28 17:45:52,515 - mu.virtual_environment:777(create_venv) INFO: Creating virtualenv: C:\Users\Larry\AppData\Local\python\mu\mu_venv-38-20210728-174449
2021-07-28 17:45:52,765 - mu.virtual_environment:778(create_venv) INFO: Virtualenv name: mu_venv-38-20210728-174449
2021-07-28 17:45:52,780 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-m', 'virtualenv', '-p', 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\Larry\\AppData\\Local\\python\\mu\\mu_venv-38-20210728-174449') with kwargs {'env': {'ALLUSERSPROFILE': 'C:\\ProgramData', 'APPDATA': 'C:\\Users\\Larry\\AppData\\Roaming', 'COMMONPROGRAMFILES': 'C:\\Program Files\\Common Files', 'COMMONPROGRAMFILES(X86)': 'C:\\Program Files (x86)\\Common Files', 'COMMONPROGRAMW6432': 'C:\\Program Files\\Common Files', 'COMPUTERNAME': 'LARRY-PC', 'COMSPEC': 'C:\\WINDOWS\\system32\\cmd.exe', 'DRIVERDATA': 'C:\\Windows\\System32\\Drivers\\DriverData', 'FP_NO_HOST_CHECK': 'NO', 'HOMEDRIVE': 'C:', 'HOMEPATH': '\\Users\\Larry', 'LOCALAPPDATA': 'C:\\Users\\Larry\\AppData\\Local', 'LOGONSERVER': '\\\\LARRY-PC', 'NUMBER_OF_PROCESSORS': '2', 'ONEDRIVE': 'C:\\Users\\Larry\\OneDrive', 'ONEDRIVECONSUMER': 'C:\\Users\\Larry\\OneDrive', 'OS': 'Windows_NT', 'PATH': 'C:\\Users\\Larry\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\PyQt5\\Qt\\bin;C:\\Program Files\\Common Files\\Microsoft Shared\\Windows Live;C:\\Program Files (x86)\\Common Files\\Microsoft Shared\\Windows Live;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Program Files (x86)\\Windows Live\\Shared;c:\\Program Files (x86)\\Microsoft SQL Server\\100\\Tools\\Binn\\;c:\\Program Files\\Microsoft SQL Server\\100\\Tools\\Binn\\;c:\\Program Files\\Microsoft SQL Server\\100\\DTS\\Binn\\;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Program Files (x86)\\QuickTime\\QTSystem\\;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files\\dotnet\\;C:\\Program Files (x86)\\dotnet\\;C:\\Users\\Larry\\anaconda3;C:\\Users\\Larry\\anaconda3\\Library\\mingw-w64\\bin;C:\\Users\\Larry\\anaconda3\\Library\\usr\\bin;C:\\Users\\Larry\\anaconda3\\Library\\bin;C:\\Users\\Larry\\anaconda3\\Scripts;C:\\Anaconda;C:\\Anaconda\\Library\\mingw-w64\\bin;C:\\Anaconda\\Library\\usr\\bin;C:\\Anaconda\\Library\\bin;C:\\Anaconda\\Scripts;C:\\Users\\Larry\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Program Files\\JetBrains\\PyCharm Community Edition 2020.3.4\\bin;;C:\\Users\\Larry\\AppData\\Local\\GitHubDesktop\\bin;C:\\Users\\Larry\\.dotnet\\tools;C:\\Users\\Larry\\AppData\\Local\\Programs\\Microsoft VS Code\\bin', 'PATHEXT': '.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC', 'PROCESSOR_ARCHITECTURE': 'AMD64', 'PROCESSOR_IDENTIFIER': 'Intel64 Family 6 Model 37 Stepping 5, GenuineIntel', 'PROCESSOR_LEVEL': '6', 'PROCESSOR_REVISION': '2505', 'PROGRAMDATA': 'C:\\ProgramData', 'PROGRAMFILES': 'C:\\Program Files', 'PROGRAMFILES(X86)': 'C:\\Program Files (x86)', 'PROGRAMW6432': 'C:\\Program Files', 'PSMODULEPATH': 'C:\\WINDO [... cut ...]
2021-07-28 18:03:00,929 - mu.logic:1523(change_mode) INFO: Workspace directory: C:\Users\Larry\mu_code
2021-07-28 18:03:02,210 - mu.logic:943(restore_session) INFO: Starting with blank file.
2021-07-28 18:03:27,685 - mu.settings:169(save) DEBUG: Saving to C:\Users\Larry\AppData\Local\python\mu\session.json
2021-07-28 18:03:27,685 - mu.logic:1387(quit) INFO: Quitting.

2021-07-28 18:03:27,685 - mu.settings:169(save) DEBUG: Saving to C:\Users\Larry\AppData\Local\python\mu\session.json
2021-07-28 18:03:27,700 - mu.settings:169(save) DEBUG: Saving to C:\Users\Larry\AppData\Local\python\mu\venv.json

carlosperate, 2021-12-17:
Thanks for the report @MrCurly2u, looks like the issue here might have been https://github.com/mu-editor/mu/issues/1755.

Assuming that's the case Mu probably run correctly the next time you tried to launch it, is that correct? Or are you still experiencing this problem?

MrCurly2u, 2021-12-18:
Did I send this report? I don't think so...

carlosperate, 2021-12-18:
Hi,

That came from the @MrCurly2u username on July 29th, did you create your account after that date?

carlosperate, 2022-01-19:
As this problem is captured in #1755, I'll close this issue to keep things tidy.
Thanks for the report!

closed 2022-01-19 by carlosperate