mu-editor/mu #1726: Crash Report

rollosson, 2021-07-31:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_GB
Platform: Windows 10 10.0.18362 AMD64

```
lf.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\VFX Prod\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\VFX Prod\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\VFX Prod\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\VFX Prod\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\VFX Prod\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 101:
Unable to create process using 'C:\Users\VFX Prod\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\VFX Prod\AppData\Local\python\mu\mu_venv-38-20210731-105621\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\VFXPRO~1\AppData\Local\Temp\tmpxo382nol\backcall-0.2.0-py2.py3-none-any.whl'

```

rollosson, 2021-07-31:
[mu.log](https://github.com/mu-editor/mu/files/6910995/mu.log)

carlosperate, 2021-08-05:
Thanks for the report @rollosson!

Is it possible you had an antivirus running while starting up Mu? If so, would you be able to disable it and try again?