mu-editor/mu #1730: Crash Report Crashes immediately after connecting m5 stack core 2 to usb

wstaniszewski, 2021-08-02:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: pl_PL
Platform: Windows 10 10.0.19041 AMD64

```
s\mu\logic.py", line 711, in check_usb
    detected = mode.find_devices(with_logging=False)
  File "C:\Users\wstan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\base.py", line 423, in find_devices
    device = self.compatible_board(port)
  File "C:\Users\wstan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\circuitpython.py", line 284, in compatible_board
    for comport in circuitpython_serial.repl_comports():
  File "C:\Users\wstan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\circuitpython_serial.py", line 69, in repl_comports
    for port in comports()
  File "C:\Users\wstan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\circuitpython_serial.py", line 49, in comports
    ports = _list_ports_windows.comports()
  File "C:\Users\wstan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\_list_ports_windows.py", line 458, in comports
    return list(iterate_comports())
  File "C:\Users\wstan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\_list_ports_windows.py", line 342, in iterate_comports
    info = list_ports_common.ListPortInfo(port_name_buffer.value, skip_link_detection=True)
TypeError: __init__() got an unexpected keyword argument 'skip_link_detection'

```

carlosperate, 2021-08-02:
Thanks for the report @wstaniszewski!
Would you be able to attach the Mu log to this GitHub Issue as well? File should be located in: `C:\Users\wstan\AppData\Local\python\mu\Logs\mu.log`

If you could first replicate this issue in your computer and then attach the log to a response to this GitHub issue it would be really useful for debugging.

wstaniszewski, 2021-08-28:
[Logs.zip](https://github.com/mu-editor/mu/files/7070265/Logs.zip)

wstaniszewski, 2021-08-28:
I hope it will work

wstaniszewski, 2021-08-28:
With files i mean  :)

carlosperate, 2021-12-17:
@wstaniszewski, I think this crash might be related to issue: #1831

We think we might have found a solution to that problem. Would you be able to try the installer at the bottom of this page, in the "Artifacts" section to see if it resolves the problem you are encountering?
https://github.com/mu-editor/mu/actions/runs/1574840282

To ensure things go smoothly it'd be highly recommended to first uninstall the version of Mu in your PC before installing this version.

If you could try this installer it would be very very useful to us to confirm the issue has been fixed.

wstaniszewski, 2021-12-18:
Hi,

Not hanging or restarting.  Hard to say if it works but it is not
crashing.   When i try to run something i got crc error2k.   But ok i
dropped this idea.

Best regards,
Wojciech Staniszewski

carlosperate, 2021-12-18:
Hi @wstaniszewski,

Thanks for getting back to us!
Does that mean you tried the test installer and that version didn't crash on startup? That would be great news!

If you are still experiencing other issues, would you be able to open new GitHub issue? That's be great to investigate that `crc error2k`.