mu-editor/mu #1731: Crash Report

barend69, 2021-08-03:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_ZA
Platform: Linux 5.4.0-1041-raspi #45-Ubuntu SMP PREEMPT Thu Jul 15 01:22:38 UTC 2021 armv7l

```
Traceback (most recent call last):
  File "/home/bob/Downloads/mu-master/mu/app.py", line 161, in run
    raise ex
  File "/home/bob/Downloads/mu-master/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/bob/Downloads/mu-master/mu/virtual_environment.py", line 621, in ensure_and_create
    self.recreate()
  File "/home/bob/Downloads/mu-master/mu/virtual_environment.py", line 566, in recreate
    self.create()
  File "/home/bob/Downloads/mu-master/mu/virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "/home/bob/Downloads/mu-master/mu/virtual_environment.py", line 878, in install_baseline_packages
    wheels.download(zipped_wheels_filepath)
  File "/home/bob/Downloads/mu-master/mu/wheels/__init__.py", line 187, in download
    pip_download(temp_dirpath, logger)
  File "/home/bob/Downloads/mu-master/mu/wheels/__init__.py", line 124, in pip_download
    raise WheelsDownloadError(
mu.wheels.WheelsDownloadError: Pip was unable to download pgzero>=1.2.1

```

barend69, 2021-08-03:
I use a raspberry pi 4 8gb
Operating system ubuntu mate 20.04.2 LTS 32-bit
mu-editor stop working after a reboot

carlosperate, 2021-08-04:
Thanks for the report @barend69!

Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

Do you have piwheels set up as a pip extra index URL? Might be done by default in Ubuntu for Raspberry Pi, but I've never used it, so I don't know: https://www.piwheels.org