mu-editor/mu #1737: Crash Report

Aditi-Rattehalli, 2021-08-05:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_AU
Platform: Windows 10 10.0.19041 AMD64

```
  self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\Navya R\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\Navya R\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\Navya R\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\Navya R\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\Navya R\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 101:
Unable to create process using 'C:\Users\Navya R\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Navya R\AppData\Local\python\mu\mu_venv-38-20210806-091103\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index "C:\Users\Navya R\AppData\Local\Temp\tmp3t6q8b62\backcall-0.2.0-py2.py3-none-any.whl"'

```

Aditi-Rattehalli, 2021-08-05:
how to login with different username

carlosperate, 2021-08-06:
Thanks for the report @Aditi-Rattehalli!

Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Navya R\AppData\Local\python\mu\Logs\mu.log`

Also, is it possible you were running an antivirus while starting up Mu? If so, would you be able to disable it and try again?