mu-editor/mu #1739: Crash Report

boodboode1, 2021-08-07:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: sv_SE
Platform: Windows 8.1 6.3.9600 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Abdisalaam\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Abdisalaam\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Abdisalaam\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\Abdisalaam\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\Abdisalaam\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 351, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\Abdisalaam\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 798, in setup
    os.makedirs(wd)
  File "C:\Users\Abdisalaam\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\Abdisalaam\\mu_code'

```

carlosperate, 2021-08-09:
Thanks for the report @boodboode1!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Abdisalaam\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-09-21:
I'll close this due to lack of activity, it looks like the race condition already tracked in #1755.

@boodboode1 if you are still experiencing this problem please reopen this issue, thanks!

closed 2021-09-21 by carlosperate