mu-editor/mu #1741: Crash Report

greenboards, 2021-08-09:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_AU
Platform: Linux 4.15.0-140-generic #144-Ubuntu SMP Fri Mar 19 14:12:35 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/murray/.local/lib/python3.6/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/murray/.local/lib/python3.6/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/murray/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 621, in ensure_and_create
    self.recreate()
  File "/home/murray/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 566, in recreate
    self.create()
  File "/home/murray/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "/home/murray/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 878, in install_baseline_packages
    wheels.download(zipped_wheels_filepath)
  File "/home/murray/.local/lib/python3.6/site-packages/mu/wheels/__init__.py", line 183, in download
    pip_download(temp_dirpath, logger)
  File "/home/murray/.local/lib/python3.6/site-packages/mu/wheels/__init__.py", line 121, in pip_download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download esptool==3.*

```

carlosperate, 2021-08-09:
Thanks for the report @greenboards!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

What distro and version are you using?
Is it possible you've pip installed with `sudo` and try to run without it?
Did you have a working internet connection when Mu was started?

greenboards, 2021-09-21:
> @greenboards <https://github.com/greenboards> Is it possible you've pip
> Would you be able to attach the Mu logs to this issue? It will really help
> us diagnose the problem.
> The file should be located in: ~/.cache/mu/mu.log

Hi

Sorry, I have re images my machine without mu editor

Regards

carlosperate, 2021-09-22:
No worries and thanks for the update.
If you can try to install Mu in the fresh system it'll also help us identify if there is an issue in your combination of OS, dependencies, etc.

carlosperate, 2021-12-17:
As the original environment that caused this issue has been reimaged I'll close this issue. But please do reopen it if you are still experiencing this problem.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate