mu-editor/mu #1743: Except KeyboardInterrupt ignored

r3dh0t, 2021-08-10:
```
try:
    # Drone code

except KeyboardInterrupt:
    # Safe land code

```

The KeyboardInterrupt exception does not run as ctrl+c kills the program regardless. My students are using Mu to program parrot mambos and I need an emergency stop for them. With Idle we'd use ctrl+c and handle the KeyboardInterrupt to land the drone safely. Unfortunatley in Mu it immediately exits the program instead, leaving the drone floating in mid-air.

carlosperate, 2021-08-10:
HI @r3dh0t,

Thanks for the report! Could you provide steps to replicate? I had a quick test and the `KeyboardInterrupt` is properly captured in Python mode:

https://user-images.githubusercontent.com/4189262/128840648-c42f23cc-c769-4e5f-ac44-9d3f18e16f59.mp4

r3dh0t, 2021-08-12:
Hi @carlosperate 
I hope this helps :)

https://user-images.githubusercontent.com/35854972/129177576-63f3983a-f695-4f18-950b-aaa07fa51348.mp4

r3dh0t, 2021-08-12:
Same behaviour unfortunately @carlosperate .

Perhaps it just occurs on Windows?
[image]

carlosperate, 2021-08-13:
You are right, I'm able to replicate this on Windows:

[image]

And it does work on Ubuntu: 
[image]

So this is a Windows specific issue.

The other thing to notice is that in macOS and Ubuntu <kbd>Ctrl + C</kbd> drops the user into the REPL, but in windows it doesn't.