mu-editor/mu #1744: Crash Report

sunbearc22, 2021-08-10:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_SG
Platform: Linux 5.4.0-80-generic #90~18.04.1-Ubuntu SMP Tue Jul 13 19:40:02 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/baymax/.local/lib/python3.6/site-packages/mu/app.py", line 161, in run
    raise ex
  File "/home/baymax/.local/lib/python3.6/site-packages/mu/app.py", line 148, in run
    venv.ensure_and_create(self.display_text)
  File "/home/baymax/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/baymax/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 768, in create
    self.install_baseline_packages()
  File "/home/baymax/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 878, in install_baseline_packages
    wheels.download(zipped_wheels_filepath)
  File "/home/baymax/.local/lib/python3.6/site-packages/mu/wheels/__init__.py", line 183, in download
    pip_download(temp_dirpath, logger)
  File "/home/baymax/.local/lib/python3.6/site-packages/mu/wheels/__init__.py", line 121, in pip_download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download esptool==3.*

```

sunbearc22, 2021-08-10:
Solved. The `setuptools_rust` python package/module was not install. Fixed using command:

`$ pip3 install --user setuptools_rust`

Then `$ mu-editor ` worked.

closed 2021-08-10 by sunbearc22

carlosperate, 2021-08-10:
Glad you resolve it @sunbearc22!

I think updating pip would have likely resolved it as well, as it would have been able to install from a wheel instead of building from the rust source code 👍