mu-editor/mu #1745: Crash Report /thanks

ermanallp, 2021-08-10:
[mu.log](https://github.com/mu-editor/mu/files/6963909/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Darwin 17.7.0 Darwin Kernel Version 17.7.0: Wed Apr 24 21:17:24 PDT 2019; root:xnu-4570.71.45~1/RELEASE_X86_64 x86_64

```
ate
    self.install_baseline_packages()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: debugpy-1.3.0-cp38-cp38-macosx_10_14_x86_64.whl is not a supported wheel on this platform.

```

carlosperate, 2021-08-11:
Hi @ermanallp, thanks for the report!

Are you in macOS 10.13 High Sierra? If so, could you try the installer from https://github.com/mu-editor/mu/issues/1683#issuecomment-877058288 ?

Thanks!

carlosperate, 2021-08-23:
I'll close this issue as being tracked in https://github.com/mu-editor/mu/issues/1683.
Thanks for the report!

closed 2021-08-23 by carlosperate