mu-editor/mu #1746: pipenv installation of mu-editor failed on Ubuntu 20.04

sunbearc22, 2021-08-10:
What should I do to overcome these installation error? 

```
$ pipenv install mu-editor
Creating a virtualenv for this project...
Pipfile: /home/user/Coding/microbit/c19sdg/Pipfile
Using /usr/bin/python3 (3.8.10) to create virtualenv...
⠸ Creating virtual environment...created virtual environment CPython3.8.10.final.0-64 in 183ms
  creator CPython3Posix(dest=/home/user/.local/share/virtualenvs/c19sdg-Ah58mGHK, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/user/.local/share/virtualenv)
    added seed packages: pip==21.2.2, setuptools==57.4.0, wheel==0.36.2
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator

✔ Successfully created virtual environment! 
Virtualenv location: /home/user/.local/share/virtualenvs/c19sdg-Ah58mGHK
Creating a Pipfile for this project...
Installing mu-editor...
Error:  An error occurred while installing mu-editor!
Error text: Collecting mu-editor
  Using cached mu_editor-1.0.3-py3-none-any.whl (2.0 MB)
Collecting pycodestyle==2.4.0
  Using cached pycodestyle-2.4.0-py2.py3-none-any.whl (62 kB)
Collecting pyqt5==5.14.1
  Using cached PyQt5-5.14.1-5.14.1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (63.5 MB)
Collecting guizero>=1.1.0
  Using cached guizero-1.2.0-py3-none-any.whl (42 kB)
Collecting PyQtChart==5.14.0
  Using cached PyQtChart-5.14.0-5.14.2-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.2 MB)
Collecting matplotlib==2.2.2
  Using cached matplotlib-2.2.2.tar.gz (37.3 MB)
Collecting semver>=2.8.0
  Using cached semver-2.13.0-py2.py3-none-any.whl (12 kB)
Collecting appdirs>=1.4.3
  Using cached appdirs-1.4.4-py2.py3-none-any.whl (9.6 kB)
Collecting gpiozero>=1.4.1
  Using cached gpiozero-1.6.2-py2.py3-none-any.whl (148 kB)
Collecting pgzero==1.2
  Using cached pgzero-1.2-py3-none-any.whl (69 kB)
Collecting qscintilla==2.11.4
  Using cached QScintilla-2.11.4-1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.9 MB)
Collecting qtconsole==4.3.1
  Using cached qtconsole-4.3.1-py2.py3-none-any.whl (108 kB)
Collecting pigpio>=1.40.post1
  Using cached pigpio-1.78-py2.py3-none-any.whl (39 kB)
Collecting requests>=2.19.1
  Using cached requests-2.26.0-py2.py3-none-any.whl (62 kB)
Collecting pyflakes==2.0.0
  Using cached pyflakes-2.0.0-py2.py3-none-any.whl (53 kB)
Collecting nudatus>=0.0.3
  Using cached nudatus-0.0.5-py3-none-any.whl (5.3 kB)
Collecting Pillow>=5.2.0
  Using cached Pillow-8.3.1-cp38-cp38-manylinux_2_5_x86_64.manylinux1_x86_64.whl (3.0 MB)
Collecting pyserial==3.4
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Collecting numpy>=1.7.1
  Using cached numpy-1.21.1-cp38-cp38-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (15.8 MB)
Collecting cycler>=0.10
  Using cached cycler-0.10.0-py2.py3-none-any.whl (6.5 kB)
Collecting pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1
  Using cached pyparsing-2.4.7-py2.py3-none-any.whl (67 kB)
Collecting python-dateutil>=2.1
  Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
Collecting pytz
  Using cached pytz-2021.1-py2.py3-none-any.whl (510 kB)
Collecting six>=1.10
  Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting kiwisolver>=1.0.1
  Using cached kiwisolver-1.3.1-cp38-cp38-manylinux1_x86_64.whl (1.2 MB)
Collecting pygame<2.0,>=1.9.2
  Using cached pygame-1.9.6.tar.gz (3.2 MB)
[... 135 lines cut ...]
    /bin/sh: 1: freetype-config: not found
    /bin/sh: 1: freetype-config: not found
    Missing dependencies
    
    Hunting dependencies...
    WARNING: "freetype-config" failed!
    SDL     : found 1.2.15
    FONT    : not found
    IMAGE   : not found
    MIXER   : not found
    PNG     : found
    JPEG    : found
    SCRAP   : found
    PORTMIDI: not found
    PORTTIME: not found
    FREETYPE: not found
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/da/b8/b110cdea95452f8f9f9e79d02062fdccaebfd808a667d6d9a805ca8d222a/pygame-1.9.2.tar.gz#sha256=2050c68ff6eba1f801efc14fd1f805aaa3a94936c80ed6725b64a75ca1bce66f (from https://pypi.org/simple/pygame/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement pygame<2.0,>=1.9.2 (from pgzero) (from versions: 1.9.2rc1, 1.9.2, 1.9.3, 1.9.4.dev0, 1.9.4rc1, 1.9.4, 1.9.5rc1, 1.9.5rc2, 1.9.5, 1.9.6rc1, 1.9.6rc2, 1.9.6, 2.0.0.dev1, 2.0.0.dev2, 2.0.0.dev3, 2.0.0.dev4, 2.0.0.dev6, 2.0.0.dev8, 2.0.0.dev10, 2.0.0.dev12, 2.0.0.dev14, 2.0.0.dev16, 2.0.0.dev18, 2.0.0.dev20, 2.0.0.dev22, 2.0.0.dev24, 2.0.0, 2.0.1.dev1, 2.0.1)
ERROR: No matching distribution found for pygame<2.0,>=1.9.2

This is likely caused by a bug in mu-editor. Report this to its maintainers.
✘ Installation Failed 

```

carlosperate, 2021-08-11:
Hi @sunbearc22,

Because the latest releases are beta you need to install with the `--pre` flag. Could you try that and let us know if it works?

Thanks!

closed 2021-08-11 by sunbearc22

carlosperate, 2021-08-12:
This should be fixed by https://github.com/mu-editor/mu/pull/1686 in the next release.

In the meantime you can install with pip instead of pipenv as I probably won't have the same lock requirements (you can still create a pipenv venv if you like using `pipenv shell` followed by `pip install --pre mu-editor`).

sunbearc22, 2021-08-12:
@carlosperate Thanks. Installation was successful. However, I could not flash to microbit due to a permission.

**Msg from mu.log.**
```
2021-08-12 15:45:02,824 - mu.modes.microbit:574(toggle_files) INFO: Toggle filesystem off.
2021-08-12 15:45:05,447 - mu.modes.microbit:273(flash) INFO: Preparing to flash script.
2021-08-12 15:45:05,448 - mu.modes.microbit:280(flash) DEBUG: Python script from 'untitled' tab:
2021-08-12 15:45:05,448 - mu.modes.microbit:281(flash) DEBUG: b'# Write your code here :-)\nfrom microbit import *\n\ndisplay.show(Image.HAPPY)\n'
2021-08-12 15:45:05,462 - mu.modes.microbit:229(find_microbit) INFO: Path to micro:bit: /media/user/MICROBIT
2021-08-12 15:45:05,463 - mu.modes.microbit:235(find_microbit) INFO: Serial port: /dev/ttyACM0
2021-08-12 15:45:05,463 - mu.modes.microbit:236(find_microbit) INFO: Device serial number: 9901000050784e450046100c000000130000000097969901
2021-08-12 15:45:05,463 - mu.modes.microbit:237(find_microbit) INFO: Board ID: 0x9901
2021-08-12 15:45:05,464 - mu.modes.microbit:358(flash) INFO: Checking target device.
2021-08-12 15:45:05,488 - mu.modes.microbit:376(flash) WARNING: Could not detect version of MicroPython.
2021-08-12 15:45:05,488 - mu.modes.microbit:423(flash_and_send) INFO: Flashing new MicroPython runtime onto device
2021-08-12 15:45:15,679 - mu.modes.microbit:459(flash_finished) INFO: Flashing successful.
2021-08-12 15:45:15,679 - mu.modes.microbit:474(copy_main) INFO: Copying main.py onto device
2021-08-12 15:45:15,679 - mu.modes.microbit:481(copy_main) INFO: ["fd = open('main.py', 'wb')", 'f = fd.write', "f(b'# Write your code here :-)\\nfrom microbit import *\\n\\ndisplay.show(')", "f(b'Image.HAPPY)\\n')", 'fd.close()']
2021-08-12 15:45:15,687 - mu.modes.microbit:497(flash_failed) ERROR: [Errno 13] could not open port /dev/ttyACM0: [Errno 13] Permission denied: '/dev/ttyACM0'
2021-08-12 15:45:15,687 - mu.interface.main:1024(show_message) DEBUG: There was a problem flashing the micro:bit.
2021-08-12 15:45:15,687 - mu.interface.main:1025(show_message) DEBUG: Please do not disconnect the device until flashing has completed. Please check the logs for more information.
```

carlosperate, 2021-08-12:
Make sure you are in the correct permissions group. More info in the note here: https://codewith.mu/en/download

I assume the same problem would happen with any other serial terminal apart from Mu.

sunbearc22, 2021-08-13:
I had previously added username in the `dialout` group before flashing. That is why I was surprised I could not flash.

```
$ sudo usermod -a -G dialout user
$ cat /etc/group | grep dialout
dialout:x:20:user
```

I also checked the serial port name for microbit and it is the same one mention in mu.log having Permission problem. 
```
$ dmesg | grep tty
[    0.083867] printk: console [tty0] enabled
[    5.989570] cdc_acm 3-6:1.1: ttyACM0: USB ACM device
[   47.167449] cdc_acm 3-6:1.1: ttyACM0: USB ACM device
[   68.225154] cdc_acm 3-6:1.1: ttyACM0: USB ACM device
[  222.276147] cdc_acm 3-6:1.1: ttyACM0: USB ACM device
[ 1553.715856] cdc_acm 3-6:1.1: ttyACM0: USB ACM device
[16725.152904] cdc_acm 3-6:1.1: ttyACM0: USB ACM device
[18254.944101] cdc_acm 3-6:1.1: ttyACM0: USB ACM device
[136066.992130] cdc_acm 3-6:1.1: ttyACM0: USB ACM device
[171929.554606] cdc_acm 3-6:1.1: ttyACM0: USB ACM device
```

After some tinkering, I discovered the serial port `/dev/ttyACM0` lacked read-write permission for others hence caused the lack of Permission problem in mu-editor. I used `sudo chmod a+rw /dev/ttyACM0 ` to fix this.

```
$ ls -l /dev/ttyACM*
crw-rw---- 1 root dialout 166, 0 Aug 13 09:33 /dev/ttyACM0
$ sudo chmod a+rw /dev/ttyACM0 
[sudo] password for user: 
$ ls -l /dev/ttyACM*
crw-rw-rw- 1 root dialout 166, 0 Aug 13 09:33 /dev/ttyACM0 
```
After this `rw` permission was given to the Others, mu-editor was able to flash to microbit.

What is annoying is that every time I unplugged and replugged microbit to Ubuntu 18.04, `mu-editor` always fail to flash to microbit because of the lack of Permission error. . I have to reissue  `sudo chmod a+rw /dev/ttyACM0 ` to fix this. I don't seem to have this issue in Ubuntu 20.04. What gives? This issue inconvenience the use of `mu-editor` especially for students. Any way around to fix this?

carlosperate, 2021-08-13:
@sunbearc22 if you could list all your finding in a new issue that would be useful, as this has now diverted enough from the original issue, and it'd be better to continue the conversation separately
Thanks for all the debugging!