mu-editor/mu #1752: Crash Report

paymerich, 2021-08-14:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 351, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 806, in setup
    os.makedirs(images_path)
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'G:\\images'

```

carlosperate, 2021-08-16:
Thanks for the report @paymerich!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\pablo\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2021-09-21:
@paymerich would you be able to attach the Mu logs to this issue? It will really help us diagnose the problem. Thanks!
File should be located in: C:\Users\pablo\AppData\Local\python\mu\Logs\mu.log

carlosperate, 2021-12-17:
Closing due to lack of activity, but please feel free to reopen if you are still encountering this issue.
You can also try the latest beta release, as it might have resolved this issue: https://github.com/mu-editor/mu/releases/

Thanks for sending the crash report!

closed 2021-12-17 by carlosperate