mu-editor/mu #1756: Crash Report

pollygecko, 2021-08-17:
I have Mu on a windows laptop, no problem, but am trying to download it on an iMac and it keeps coming up with this issue:
I also tried downloading it through pip3 install in the terminal, but apparently it can't find version 1.1.0b5. (Tried using an older version, but then comes up with some error about matplotlib not being supported, no matter which version I try). I need this for an internship, please help! thank you

**REMEMBER TO ATTACH YOUR LOG FILE**
[mu.log](https://github.com/mu-editor/mu/files/7002030/mu.log)
[mu.log](https://github.com/mu-editor/mu/files/7002034/mu.log)

Version: 1.1.0.beta.5
Locale: en_US
Platform: Darwin 17.7.0 Darwin Kernel Version 17.7.0: Fri Oct 30 13:34:27 PDT 2020; root:xnu-4570.71.82.8~1/RELEASE_X86_64 x86_64

```
ate
    self.install_baseline_packages()
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/Applications/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: debugpy-1.3.0-cp38-cp38-macosx_10_14_x86_64.whl is not a supported wheel on this platform.

```

carlosperate, 2021-08-17:
Hi @pollygecko, thanks for the report!

Are you in macOS 10.13 High Sierra? This is a known issue with one of the packed dependencides. Could you try the installer from https://github.com/mu-editor/mu/issues/1683#issuecomment-877058288 ?

Thanks!

pollygecko, 2021-08-17:
Hi @carlosperate,

Yes I'm on version 10.13.6. I just tried the installer from the other comment, it worked! Thanks so much for your help :)

closed 2021-08-17 by pollygecko