mu-editor/mu #1757: Crash Report

bara81, 2021-08-18:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.5
Locale: en_US
Platform: Darwin 17.7.0 Darwin Kernel Version 17.7.0: Thu Jun 18 21:21:34 PDT 2020; root:xnu-4570.71.82.5~1/RELEASE_X86_64 x86_64

```
_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/private/var/folders/p3/v45mz08x4nb8pnrzz810vq6w0000gn/T/AppTranslocation/3042649B-A290-4CB4-8557-9CC92927C152/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/private/var/folders/p3/v45mz08x4nb8pnrzz810vq6w0000gn/T/AppTranslocation/3042649B-A290-4CB4-8557-9CC92927C152/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/private/var/folders/p3/v45mz08x4nb8pnrzz810vq6w0000gn/T/AppTranslocation/3042649B-A290-4CB4-8557-9CC92927C152/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/private/var/folders/p3/v45mz08x4nb8pnrzz810vq6w0000gn/T/AppTranslocation/3042649B-A290-4CB4-8557-9CC92927C152/d/Mu Editor.app/Contents/Resources/Python/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: debugpy-1.3.0-cp38-cp38-macosx_10_14_x86_64.whl is not a supported wheel on this platform.

```

carlosperate, 2021-08-18:
Hi @bara81, thanks for the report and the log!

Are you in macOS 10.13 High Sierra? This is a known issue with one of the packed dependencies. Could you try the installer from https://github.com/mu-editor/mu/issues/1683#issuecomment-877058288 ?

Thanks!

bara81, 2021-08-19:
> packed dependencies. Could you try the installer from #1683 (comment)

Hi

Thanks, everything seems fine👍

Best regrads,
Alex

carlosperate, 2021-08-19:
Awesome, glad that worked!

closed 2021-08-19 by carlosperate