mu-editor/mu #1760: Crash Report

Gilnaa, 2021-08-19:
[mu.log](https://github.com/mu-editor/mu/files/7015599/mu.log)

Version: 1.1.0.beta.5
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
s(zipped_wheels_filepath)
  File "C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 855, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 101:
Unable to create process using 'C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Gilad Naaman\AppData\Local\python\mu\mu_venv-38-20210819-175625\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\GILADN~1\AppData\Local\Temp\tmp59dc_f5p\backcall-0.2.0-py2.py3-none-any.whl'

```

carlosperate, 2021-08-19:
Hi @Gilnaa, thanks for the report and log!

Is it possible you are running an antivirus while starting up Mu? If so, would you be able to disable it and try again?

Gilnaa, 2021-08-19:
Oh, good point.
Windows Defender is running

Gilnaa, 2021-08-19:
Seems to persist even when windows defender is disabled (also tried running as administrator)

Gilnaa, 2021-08-19:
Weirdly enough `backcall` is already installed:
```
C:\Users\Gilad Naaman>"C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip install backcall
Requirement already satisfied: backcall in c:\users\gilad naaman\appdata\local\programs\mu editor\python\lib\site-packages (0.2.0)
```

carlosperate, 2021-08-20:
Thanks for testing @Gilnaa!

In this case Mu is failing when trying to install wheels in a newly created virtual environment, which is later used to run the user code from the editor.

The command that is failing in this case would be this one, using the pip executable from the venv:
```
"C:\Users\Gilad Naaman\AppData\Local\python\mu\mu_venv-38-20210819-175625\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\GILADN~1\AppData\Local\Temp\tmp59dc_f5p\backcall-0.2.0-py2.py3-none-any.whl'
``` 

However, because this is a temporary file (all wheels are zipped for packaging, and then unzip during the installation process), I doubt you'd be able to run it as it is.

I think the main problem is that Python is not being able to launch the process in itself, so probably the pip command wouldn't have issues installing the dependency, but it doesn't get that far.

We've seen this issue a few times already, and unfortunately we haven't been able to replicate yet in our environments.
@Gilnaa if you are up for a challenge to debug this it would be greatly appreciated, I'm happy to provide all the guidance required.

Gilnaa, 2021-08-20:
Sure, I have some free time this weekend, tell me what I can do.

carlosperate, 2021-08-20:
That is fantastic news, thank you so much @Gilnaa! 

- First I would try to run Mu from source code, in a new virtual environment: https://mu.readthedocs.io/en/latest/setup.html
    - If it runs correctly maybe it's a problem with the version of the Python and/or dependencies packaged in the installer
- If source code works, then I would try to replicate this with the latest version of the CI installer: https://github.com/mu-editor/mu/actions/runs/1125019808
    - Similarly, this CI installer might have newer versions of stuff, so it'd be interesting to know as well.
    - It the CI installer works, make sure you uninstall it and install beta.5 again, to be able to replicate the issue
- If neither installer works, but running from source code does work, then I would try to build the installer yourself and see if that installation works. If you've installed the dev dependencies (`pip install -e ."[dev]"`), then the installer can be built with `python make.py win64`
- This is going to sound weird, but we also found some similar issues when the Windows user name had a space in them. Would you also be able to create a new user in your PC without a space in the user name? Then install beta.5 in that new user and see if it works?

If none of this works I would try to run the `"C:\...\pip.exe" install ... backcall-0.2.0-py2.py3-none-any.whl` command manually while the temp files are still present on your computer. From the log file you can see the command that Mu tried to execute via a subprocess:

```
2021-08-19 17:56:25,867 - mu.virtual_environment:850(install_from_zipped_wheels) INFO: About to install from wheel: backcall-0.2.0-py2.py3-none-any.whl
2021-08-19 17:56:25,867 - mu.virtual_environment:114(run_blocking) INFO: About to run blocking C:\Users\Gilad Naaman\AppData\Local\python\mu\mu_venv-38-20210819-175625\scripts\pip.exe with args ['install', '--disable-pip-version-check', '--no-deps', '--no-index', 'C:\\Users\\GILADN~1\\AppData\\Local\\Temp\\tmp59dc_f5p\\backcall-0.2.0-py2.py3-none-any.whl'] and envvars {}
2021-08-19 17:56:25,968 - mu.virtual_environment:164(wait) DEBUG: Finished: True; exitStatus 0; exitCode 101
2021-08-19 17:56:25,968 - mu.virtual_environment:185(wait) ERROR: Unable to create process using 'C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Gilad Naaman\AppData\Local\python\mu\mu_venv-38-20210819-175625\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\GILADN~1\AppData\Local\Temp\tmp59dc_f5p\backcall-0.2.0-py2.py3-none-any.whl'
2021-08-19 17:56:25,979 - mu.virtual_environment:643(ensure_and_create) ERROR: Process finished but with error code 101:
Unable to create process using 'C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Gilad Naaman\AppData\Local\python\mu\mu_venv-38-20210819-175625\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\GILADN~1\AppData\Local\Temp\tmp59dc_f5p\backcall-0.2.0-py2.py3-none-any.whl'
```

If the temp files disapear before you can run the command you could just get the wheels zip file from the Mu installation files, unzip it somewhere (maybe inside the temp directory would be better) and use that path.
The zip file with the wheels should be in `C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\Lib\site-packages\mu\wheels\`

Thank you again for your help debugging this issue, it's truly appreciated! 🌟

carlosperate, 2021-08-20:
Also, could you run `pip list` with the Mu python interpreter (it packs it's own Python) and the created venv?

```
"C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list
```

```
"C:\Users\Gilad Naaman\AppData\Local\python\mu\mu_venv-38-20210819-175625\scripts\pip.exe" list
```

And also, in case this is a weird bug between running `python.exe` vs `pythonw.exe` (the `w` is for windowed mode, and some things are different). Could you try to run Mu from a command prompt like this and see if it success?

```
"C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\python.exe" -m mu
```

You could try everything from this message first, before the other things I suggested. Thanks again!

Gilnaa, 2021-08-20:
Works from souce but not CI installer.
Created another user without space in the name and as you said the installer worked :)

Back to the original user:
I cannot run `pip.exe` (or `python` for that matter) from the directory of the mu venv, I keep getting this vague "Unable to create process using 'C:\Users...'" error.

Error persists even when running from console as your last message suggested.

carlosperate, 2021-08-23:
Thanks @Gilnaa! 

> Created another user without space in the name and as you said the installer worked :)

This is really interesting, thanks for testing this. I wonder why this is an issue with the packaged installer, but it is fine when running from source code.

Would you be able to create an installer from source and see if that version works?

I've been trying to create users on Windows with a space in the name and I still haven't been able to replicate. I wonder if it might be a red herring and be unrelated.
Would you be able to create a new account, but this time with a space to see if beta.5 fails there?
Maybe the issues is to do with something else in the environment.

> I cannot run pip.exe (or python for that matter) from the directory of the mu venv, I keep getting this vague "Unable to create process using 'C:\Users...'" error.

Oh right, so running a simple pip command, from the venv, like this one failed with a `Unable to create process ` error?
```
"C:\Users\Gilad Naaman\AppData\Local\python\mu\mu_venv-38-20210819-175625\scripts\pip.exe" list
```

Would you be able to copy/paste the command used and error? (path used and printed in the error might be important)

Did you also get the same issue using the python.exe?

```
"C:\Users\Gilad Naaman\AppData\Local\python\mu\mu_venv-38-20210819-175625\scripts\python.exe" -m pip list
```

Gilnaa, 2021-08-23:
Created a user `boaty mcboatface` and it worked fine.
I created an installer from source and it failed as well on the main user.

Something odd I noticed:

I had Python3.9 installed as system python in `C:\Program Files\...` and could create `venv`s just fine with it.
Couldn't create an installer with it because of some assertions in `mu`, so I installed Python3.8 just for my user, (`C:\Users\Gilad ...\AppData\Local\...`) and tried to create a venv with it.
Creating the venv failed because it couldn't run pip (no details)

After creating the venv using `--without-pip`, I activated it and saw that I cannot run `python`, getting another `Unable to create process`.
I have a strong feeling that you're right, it's not about the white-space.
I don't think it's actively related to an antivirus since I turned off Windows Defender (and, well, installers work on other users).

My guess would be that my homedir is somehow "marked" in a way that makes windows dislike venv created from non-system python.
It's pretty hard to think of what to do further though, since I mostly do programming on Linux and Windows is a blackbox to me.

As for running the MU venvs:
[image]

carlosperate, 2021-08-23:
That is very useful, thanks @Gilnaa!

>Created a user boaty mcboatface and it worked fine.
>I created an installer from source and it failed as well on the main user.

Okay, this is great progress. So we know that:
- This not related to the spaces in the user name
- It is replicated running from source
- It is only an issue when running from the installer

I've installed Python 3.9 as the system Python in `C:\Program Files\...`, and the 3.8 only for my users as well, but I am still unable to replicate.

It could be related to this: https://bugs.python.org/issue37369

When you installed the system Python did you use the Windows store version or an installer from the Python website?
Do you have any non-ascii characters on your username (doesn't look like it).
Have you enabled case sensitivity on Windows? Do these two commands work the same?
```
"C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\python.exe" --version
```
```
"C:\Users\Gilad Naaman\AppData\Local\Programs\mu editor\Python\python.exe" --version
```

-----

I assume this fails the same way as you mentioned in your last post?
```
"C:\Users\Gilad Naaman\AppData\Local\python\mu\mu_venv-38-20210819-175625\scripts\python.exe" --version
```

What do you get on this command?

```
"C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.executable)"
```

Could you also run this to see if the `op.txt` file is created and what it is written to it?
```
"C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\pythonw.exe" -c "import sys; open('op.txt', 'w').write('{}\n{}'.format(sys.executable, sys._base_executable))"
```

The "obvious" answer would be "C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\pythonw.exe" twice, but if it points somewhere else it could be the reason the venvs have corrupted executables.

Gilnaa, 2021-08-23:
Oh, I might have phrased it badly. I told the 3.8/3.9 thing to just explain how I got there.
This bug also reproduces if I install Python3.9 just for my user, I don't think it's related to the Python version, maybe it's just venv's reaction to the install location.

My username only uses ascii AFAICT, unless microsoft inserted some ZWJ or something funky like that.

I've installed Python using an installer from python.org.

Both versions work, it looks like the FS is still case-insensitive.

Yes, calling the python exec inside the venv fails the same way:
```
C:\Users\Gilad Naaman>"C:\Users\Gilad Naaman\AppData\Local\python\mu\mu_venv-38-20210819-175625\scripts\python.exe" --version
Unable to create process using 'C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\python.exe --version'
```

```
C:\Users\Gilad Naaman>"C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\python.exe" -c "import sys; print(sys.executable)"
C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\python.exe
```

```
C:\Users\Gilad Naaman>"C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\pythonw.exe" -c "import sys; open('op.txt', 'w').write('{}\n{}'.format(sys.executable, sys._base_executable))"

C:\Users\Gilad Naaman>type op.txt
C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\pythonw.exe
C:\Users\Gilad Naaman\AppData\Local\Programs\Mu Editor\Python\pythonw.exe

> Once again, thank you so much with all your help with this, it's been truly valuable!
No problem at all :)
Glad to be of service, and honestly pretty curious
```